openwrt/packages #5351: php7 - imagemagick module?

eliz82, 2018-01-04:
Maintainer: @Maintainer: @mhei ?
Environment: Entware / Voxel Firmware / Netgear R7800

Description:
any chance to get the imageck (imagemagick) PHP module? https://github.com/mkoppanen/imagick
it seem they made it compatible with php7 starting with imageck 3.4.x version.

this module is more more future rich then php GD module and more better then using external imagemagick using PHP "exec" function

pprindeville, 2018-01-04:
Why would anyone need this on an embedded firewall/router?

jow-, 2018-01-04:
Charting, traffic stats, pie charting quotas, ...

The same could bes asked about PHP ;)

pprindeville, 2018-01-04:
Fair enough.

mhei, 2018-01-04:
First step should be to (re-)integrate the library because the pecl extension would need it as dependency when I see this correctly. When the library is added, then we can talk about packaging this new pecl extension and I can take maintainership of the new pecl package, so I'll assign this issue to myself, to keep it on my radar. But I do not intend to become maintainer of imagemagick library, so volunteers are welcome :smile:

mhei, 2018-07-29:
I'm closing this issue: as long as nobody wants to maintain imagemagick library for OpenWrt, we do not have package requirements satisfied.

closed 2018-07-29 by mhei

flyn-org, 2020-12-21:
@crisc82 and @mhei, please see https://github.com/openwrt/packages/pull/14286. This is useful for running NextCloud on OpenWrt.