openwrt/packages #5361: python,python3: removal of .egg-info disrupts Python package discovery

jefferyto, 2018-01-06:
Maintainer: @commodo
Environment: OpenWRT/LEDE trunk

Description:
According to the [setuptools documentation](https://setuptools.readthedocs.io/en/latest/formats.html#eggs-and-their-formats), the `.egg-info` file/directory contains the metadata for each Python project/package. Without this information, programs cannot use the [`pkg_resources`](https://setuptools.readthedocs.io/en/latest/pkg_resources.html) module to discover installed packages.

For example, the error in #5030 is caused by a missing `.egg-info` directory. `/usr/bin/obfsproxy` is a short Python script that calls [`load_entry_point()`](https://setuptools.readthedocs.io/en/latest/pkg_resources.html#convenience-api) to load the actual code from the obfsproxy project/module.

You could say I (as the maintainer of obfsproxy package) should just add `PYTHON_KEEP_EGGINFO` to the obfsproxy Makefile, but I would argue that `.egg-info`'s should not be removed by default. If I was a normal user and wrote a Python program for my OpenWRT/LEDE device that uses the `pkg_resources` module, I shouldn't have to recompile every Python package I want to use just to get it to work.

jefferyto, 2018-01-06:
As a side note, the `.egg-info` file/directory also contains dependency information (in `requires.txt`). Once I copied the `.egg-info` for obfsproxy back into the correct place, the next error I get is:

```
pkg_resources.DistributionNotFound: The 'pyyaml' distribution was not found and is required by obfsproxy
```

because it couldn't find the `.egg-info` for pyyaml (python-yaml in OpenWRT/LEDE).

This dependency information is probably redundant. I looked at the `.egg-info`'s installed on my Ubuntu machine, and each has had their core requirements removed from `requires.txt`. I believe the Debian packaging system removes this information when it converts them into .deb dependencies (but I haven't pinpointed where this happens yet).

commodo, 2018-01-07:
So, you could try PYTHON_KEEP_EGGINFO / PYTHON3_KEEP_EGGINFO, from this commit here: https://github.com/openwrt/packages/commit/074d2863be48c1a9093fffcda91f7a328bc7d0cb

But, if there are 2-3 more packages that need this, maybe it makes sense to just remove the whole egg-info removal logic.

jefferyto, 2018-01-07:
I'm saying the whole egg-info removal logic should be removed. Here is a program I just wrote:

```python
from pkg_resources import working_set
for dist in working_set:
    print(dist)
```

On a 17.01.4 VM where I installed obfsproxy, this is the output:

```
wsgiref 0.1.2
Python 2.7.14
pycrypto 2.6.1
pyptlib 0.0.6
setuptools 27.2.0
PyYAML 3.11
zope.interface 4.3.2
Twisted 16.4.1
obfsproxy 0.2.13
```

On a trunk VM with obfsproxy installed, this is the output:

```
pycrypto 2.6.1
```

The trunk output should match the 17.01.4 output.

commodo, 2018-01-07:
Fine by me.
Will create a PR to remove it :)

jefferyto, 2018-01-08:
Thanks, I've opened a PR #5376

Also thanks for "adopting" some of my packages :smile:

commodo, 2018-01-08:
no worries :)
i also did it to try out [and develop] some of the `PyBuildPackage` rules to help support both Py2 & Py3

PR #5376 (2018-01-08): python,python3: undo egg-info removal
Maintainer: @commodo
Compile tested: ar71xx, OpenWRT/LEDE trunk
Run tested: none

Description:
This reverts commits 4333d1dcbfd67dde4ca7d08717bc345cc424fbf8 and
074d2863be48c1a9093fffcda91f7a328bc7d0cb, making Python packages
discoverable again by pkg_resources.

Fixes #5361.
commit c5cbd44c (2018-01-08): python,python3: undo egg-info removal

This reverts commits 4333d1dcbfd67dde4ca7d08717bc345cc424fbf8 and
074d2863be48c1a9093fffcda91f7a328bc7d0cb, making Python packages
discoverable again by pkg_resources.

Fixes #5361.
closed 2018-01-08 by hnyman