openwrt/packages #5362: python-pyopenssl: error on import OpenSSL in python2.7

valdi74, 2018-01-06:
Maintainer: Jeffery To <jeffery.to@gmail.com>, Alexandru Ardelean <ardeleanalex@gmail.com>
Environment: (Xiaomi Mi Router 3G, OpenWrt SNAPSHOT, r5682-b153dbf)

Description:

There is an error 'Error relocating /usr/lib/python2.7/site-packages/cryptography/hazmat/bindings/_openssl.so: EC_GROUP_get_curve_GF2m: symbol not found' when trying to import module OpenSSL in python:

```
root@OpenWrt:~# python
Python 2.7.14 (default, Jan  6 2018, 16:59:31) 
[GCC 5.5.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import OpenSSL
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python2.7/site-packages/OpenSSL/__init__.py", line 8, in <module>
  File "/usr/lib/python2.7/site-packages/OpenSSL/crypto.py", line 16, in <module>
  File "/usr/lib/python2.7/site-packages/OpenSSL/_util.py", line 6, in <module>
  File "/usr/lib/python2.7/site-packages/cryptography/hazmat/bindings/openssl/binding.py", line 13, in <module>
ImportError: Error relocating /usr/lib/python2.7/site-packages/cryptography/hazmat/bindings/_openssl.so: EC_GROUP_get_curve_GF2m: symbol not found
```

commodo, 2018-01-11:
hmm ; i wonder how i missed this
pyopenssl seems a bit sensitive [breaks easily]
will take a look ; but it looks like openssl related ;

jefferyto, 2018-03-22:
I tried reproducing this today (armvirt-32 on qemu, 8 March snapshot; python packages 21 March snapshot), and I didn't get the error on import. Is this issue still present?

valdi74, 2018-03-22:
On Xiaomi Mi Router 3G OpenWrt SNAPSHOT, r6427-ed3860c from Mar 13 problem still exists:
```
root@OpenWrt:~# python
Python 2.7.14 (default, Mar 13 2018, 21:39:24)
[GCC 7.3.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import OpenSSL
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python2.7/site-packages/OpenSSL/__init__.py", line 8, in <module>
  File "/usr/lib/python2.7/site-packages/OpenSSL/crypto.py", line 16, in <module>
  File "/usr/lib/python2.7/site-packages/OpenSSL/_util.py", line 6, in <module>
  File "/usr/lib/python2.7/site-packages/cryptography/hazmat/bindings/openssl/binding.py", line 13, in <module>
ImportError: Error relocating /usr/lib/python2.7/site-packages/cryptography/hazmat/bindings/_openssl.so: EC_GROUP_get_curve_GF2m: symbol not found

```

but on fresh install x86_64 OpenWrt SNAPSHOT, r6515-3db9d6e57d from Mar 21 there is no issue:
```
root@OpenWrt:~# python
Python 2.7.14 (default, Mar 21 2018, 22:24:09)
[GCC 7.3.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import OpenSSL
>>>
```

I will test it on real hardware (Mi Router 3G) when I get it in my hands (few days).

valdi74, 2018-03-22:
On Xiaomi Mi Router 3G OpenWrt SNAPSHOT, r6518-8bb9f8dd47 from Mar 21 there is no issue. Closing.

closed 2018-03-22 by valdi74