openwrt/packages #5368: unbound: domain_insecure setup via uci is broken

silentcreek, 2018-01-07:
Maintainer: @EricLuehrsen 
Environment: LEDE 17.01.4, x86

I added a list of domains to the unbound configuration via uci, but then I noticed that they do not end up in the autogenerated configuration. I looked at the source code of unbound.sh and I think I found the culprit there:

Basically, the function unbound_conf sets the variable cfg to the first positional parameter, see [1]. But unbound_conf is not called with any argument (see [2]), so it'll always be empty. The variable cfg is required, however, when the domain_insecure list is processed (see [3]), otherwise config_list_foreach will not return anything.

Hence, I think the code should be fixed in a way that unbound_conf will be invoked with the corresponding uci config section variable. The code is the same for master and 17.01, so I assume both need to be fixed.

[1] https://github.com/openwrt/packages/blob/master/net/unbound/files/unbound.sh#L571
[2] https://github.com/openwrt/packages/blob/master/net/unbound/files/unbound.sh#L1162
[3] https://github.com/openwrt/packages/blob/master/net/unbound/files/unbound.sh#L866

EricLuehrsen, 2018-01-08:
Solved in #5374 for trunk. Not a security bug, no other reported user complaints (12 months), and possible work around in `unbound_srv.conf`, so likely does not meet the bar for back port in 17.1.4. If acceptable, then please close ticket.

silentcreek, 2018-01-08:
Thanks for the quick response and fix. Personally, I don't need a backport as I already worked around it by editing unbound_srv.conf, as you suggest, too. However, in this case, I would recommend updating the documentation/readme to reflect that domain_insecure via uci is not supported on LEDE 17.01.

PR #5374 (2018-01-09): unbound: fix uci for option domain-insecure:
Maintainer: me
Tested: TL-Archer-C7v2 trunk
Description: Fixes #5368 -- Unbound UCI option mapped to domain-insecure:
closed 2018-01-08 by silentcreek