openwrt/packages #5379: dnscrypt-proxy: Missing repository and replacement/alternative package

sditalan12, 2018-01-08:
Maintainer: @damianorenfer 
Environment: LEDE 17.01.4 MR3220 v2

Description:
Several days ago, dnscrypt-proxy repo from jedisct1 are removed and thus I can't build the package because of this. Apparently, there is a better solution which is using DNS-over-TLS although there are only few dns servers that are supporting this method.

thess, 2018-01-08:
It looks like the project has been (at least temporarily) moved to [dyne.org](https://github.com/dyne/dnscrypt-proxy) for support and maintenance. I will update the package to use this new repo shortly unless someone else does it first. dnscrypt-resolvers.csv location has changed also.

thess, 2018-01-08:
Package updated...

closed 2018-01-08 by thess