openwrt/packages #5380: python,python3: cannot generate -src packages for other Python modules

jefferyto, 2018-01-08:
Maintainer: @commodo 
Environment: OpenWRT/LEDE trunk

Description:
I'm trying to add `-src` packages for other modules but I'm getting an error.

If I edit `lang/python/python-ply/Makefile` and add these lines at the end:

```
$(eval $(call BuildPackage,python-ply-src))
$(eval $(call BuildPackage,python3-ply-src))
```

and then do `./scripts/feeds update -a`, I get this error:

```
ERROR: please fix feeds/packages/lang/python/python-ply/Makefile - see logs/feeds/packages/lang/python/python-ply/dump.txt for details
```

Inside that log file is this:

```
Package: python-ply
Submenu: Python
Version: 3.10-1
Depends: +libc +SSP_SUPPORT:libssp +USE_GLIBC:librt +USE_GLIBC:libpthread +PACKAGE_python-ply:python-light
Conflicts: 
Menu-Depends: 
Provides: 
Build-Variant: python
Section: lang
Category: Languages
Repository: base
Title: python-ply
Maintainer: Jeffery To <jeffery.to@gmail.com>
Source: ply-3.10.tar.gz
License: BSD-3-Clause
LicenseFiles: README.md
Type: ipkg
Description: PLY is a 100% Python implementation of the common parsing tools lex
and yacc.
http://www.dabeaz.com/ply/
Jeffery To <jeffery.to@gmail.com>
@@

Package: python3-ply
Submenu: Python
Version: 3.10-1
Depends: +libc +SSP_SUPPORT:libssp +USE_GLIBC:librt +USE_GLIBC:libpthread +PACKAGE_python3-ply:python3-light
Conflicts: 
Menu-Depends: 
Provides: 
Build-Variant: python3
Section: lang
Category: Languages
Repository: base
Title: python3-ply
Maintainer: Jeffery To <jeffery.to@gmail.com>
Source: ply-3.10.tar.gz
License: BSD-3-Clause
LicenseFiles: README.md
Type: ipkg
Description: PLY is a 100% Python implementation of the common parsing tools lex
and yacc.
.
(Variant for Python3)
http://www.dabeaz.com/ply/
Jeffery To <jeffery.to@gmail.com>
@@

Package: python-ply-src
Version: 3.10-1
Depends: +libc +SSP_SUPPORT:libssp +USE_GLIBC:librt +USE_GLIBC:libpthread 
Conflicts: 
Menu-Depends: 
Provides: 
Section: opt
Category: Extra packages
Repository: base
Title: 
Maintainer: Jeffery To <jeffery.to@gmail.com>
Source: ply-3.10.tar.gz
License: BSD-3-Clause
LicenseFiles: README.md
Type: ipkg
Description: 
Jeffery To <jeffery.to@gmail.com>
@@

Makefile:68: *** Package/python-ply-src is missing the TITLE field.  Stop.
```

I've tried tinkering with `python-package.mk` but I haven't had any success so far.

commodo, 2018-01-08:
can you show me a full Makefile ?

jefferyto, 2018-01-08:
```
#
# Copyright (C) 2015-2016 OpenWrt.org
#
# This is free software, licensed under the GNU General Public License v2.
# See /LICENSE for more information.
#

include $(TOPDIR)/rules.mk

PKG_NAME:=python-ply
PKG_VERSION:=3.10
PKG_RELEASE:=1

PKG_SOURCE:=ply-$(PKG_VERSION).tar.gz
PKG_SOURCE_URL:=http://www.dabeaz.com/ply
PKG_HASH:=96e94af7dd7031d8d6dd6e2a8e0de593b511c211a86e28a9c9621c275ac8bacb

PKG_BUILD_DIR:=$(BUILD_DIR)/$(BUILD_VARIANT)-ply-$(PKG_VERSION)

PKG_LICENSE:=BSD-3-Clause
PKG_LICENSE_FILES:=README.md
PKG_MAINTAINER:=Jeffery To <jeffery.to@gmail.com>

include $(INCLUDE_DIR)/package.mk
$(call include_mk, python-package.mk)
$(call include_mk, python3-package.mk)

PKG_UNPACK:=$(HOST_TAR) -C $(PKG_BUILD_DIR) --strip-components=1 -xzf $(DL_DIR)/$(PKG_SOURCE)

define Package/python-ply/Default
  SECTION:=lang
  CATEGORY:=Languages
  SUBMENU:=Python
  URL:=http://www.dabeaz.com/ply/
endef

define Package/python-ply
$(call Package/python-ply/Default)
  TITLE:=python-ply
  DEPENDS:=+PACKAGE_python-ply:python-light
  VARIANT:=python
endef

define Package/python3-ply
$(call Package/python-ply/Default)
  TITLE:=python3-ply
  DEPENDS:=+PACKAGE_python3-ply:python3-light
  VARIANT:=python3
endef

define Package/python-ply/description
PLY is a 100% Python implementation of the common parsing tools lex
and yacc.
endef

define Package/python3-ply/description
$(call Package/python-ply/description)
.
(Variant for Python3)
endef

$(eval $(call PyPackage,python-ply))
$(eval $(call BuildPackage,python-ply))

$(eval $(call Py3Package,python3-ply))
$(eval $(call BuildPackage,python3-ply))

$(eval $(call BuildPackage,python-ply-src))
$(eval $(call BuildPackage,python3-ply-src))
```

commodo, 2018-01-08:
maybe try:

```
$(eval $(call PyPackage,python-ply))
$(eval $(call BuildPackage,python-ply))
$(eval $(call BuildPackage,python-ply-src))

$(eval $(call Py3Package,python3-ply))
$(eval $(call BuildPackage,python3-ply))
$(eval $(call BuildPackage,python3-ply-src))
```

commodo, 2018-01-08:
but if that works, then it's a bug in the logic of the python sources stuff

jefferyto, 2018-01-08:
Same error occurs

commodo, 2018-01-08:
hmm ; i think `python-package.mk` is not present in the STAGING_DIR/mk folder

which [if true] also means that it's a bit hard to package Python source files outside of Python/Python3

can you check for  `python-package.mk` ?

jefferyto, 2018-01-08:
`python-package.mk` exists in `staging_dir/target-.../mk`, but the feeds script appears to use the version in `python/files`. I changed the "(sources)" title text in both `python-package.mk` files to different text, and the generated package feed picks up the altered text from the version in `python/files`.

commodo, 2018-01-08:
i'll try out tomorrow this patch: https://github.com/commodo/packages/commit/08fb7920206eb03a9cbea14df4b58a505181899c

it may be a bit hard to swallow as is, but I don't have a better idea atm
it may not be a bad idea in general ; since it would localize the `python-package.mk` to the packages feed

another idea, would be to move all the mk files the `lang/python/` folder directly
the `include_mk` function is part of OpenWrt and Python is the only user of it
i would not depend on core OpenWrt if possible for Python-only stuff ;

NeoRaider, 2018-01-10:
I've tested your patch commodo/packages@08fb792, and found that there is still the issue of the *include_mk* inside python-host.mk and python-package.mk (and their python3 versions), which leads to the issue described in #5401 when switching targets.

I have found a solution that seems to work and pushed it to https://github.com/NeoRaider/openwrt-packages/commits/python_mk together with your patch.

commodo, 2018-01-10:
@NeoRaider 
yeah ; so, my thought about this is to not install the mk code snippets ;
they are an inheritance from Python since before the packages feed was hosted on Github, and I never thought to wonder about them [ since they just worked ]

but now [since I know the build system better], I figured it was a terrible idea to begin with [also the whole `include_mk` construct]
which is why i closed PR https://github.com/openwrt/openwrt/pull/590 and I'll move to removing the `include_mk` completely

NeoRaider, 2018-01-10:
@commodo Does my branch look okay to you? If it does, I can push it and will then remove `include_mk` from the base repo.

commodo, 2018-01-10:
I've been working on this a bit today.
I took some ideas from your branch and put them here: https://github.com/commodo/packages/commits/python-package-mk-localize

i haven't managed to test this yet;
[ i'm load balancing other stuff ]
the gist of this, is that I think we should move the mk files outside of `lang/python/python` and `lang/python/python3` directly to `lang/python` ; to make the somewhat "exported" and usable by other packages ;

also, i've been thinking a bit about packages outside of this feed ;
example here: https://github.com/openwrt/telephony/blob/master/net/freeswitch-stable/Makefile#L280

i think a similar approach as to perl `FS_STABLE_PERL_FEED:=$(TOPDIR)/feeds/packages/lang/perl`  would probably work for those cases ; it's not perfect, but it is a bit better than `include_mk`

another idea, would to maybe re-work `include_mk` to search for `mk` files in feeds;
that would mean that feeds would need to define a `exports` section ;
but maybe that's a bit much for now

in any case, I would like to move the mk files outside of the python folders
if you have time to take a look at my branch, feel free to do so and take those commits ;
if not, i will try to test them and come back

commodo, 2018-01-10:
@NeoRaider 
opened PR https://github.com/openwrt/packages/pull/5419

so, in the next few days I'll try to monitor for any break-age.
feel free to ping me about it, in case i miss it ;

this weekend i will be AFK, but hopefully i can catch any by then ; if not, next week

commodo, 2018-01-11:
@jefferyto 
can you retry with the merged PR/code

jefferyto, 2018-01-11:
Seems to be working now - thanks!

closed 2018-01-11 by jefferyto