openwrt/packages #5405: Git package missing templates directory

gmerrall, 2018-01-10:
Maintainer: @<github-user> (find it by checking history of the package Makefile)
Environment: arm

Description:
The git templates template directory is not populated which causes issues when creating git repositories. Normally when you run "git init --bare" a number of files are copied over including the contents of the 'template' directory which typically contains sample scripts.

Perform the following steps on a regular box with git installed

$ strings which git | grep /templates
share/git-core/templates
$ ls /usr/share/git-core/templates/
branches	description	hooks	info

If this is done with an openwrt  install

$ strings which git | grep /templates
/opt/share/git-core/templates
$ ls /opt/share/git-core/templates
<empty>

Expected result is the template files should be present

You can also do the same action by running 'git init --bare' and observe different results.

gmerrall, 2018-01-10:
Maintainer: @tripolar

tripolar, 2018-01-11:
would be nice to provide a patch with the change next time.
but this is trivial, thanks for reporting - fixed in https://github.com/openwrt/packages/commit/f3490fe273fba4d113804819add17b3e870f28d7

closed 2018-01-11 by tripolar