openwrt/packages #5406: bluez agent in bluez 5.37

PraneetKaur, 2018-01-10:
Hi, 
I would like to know if anyone has ported agent functionality in the bluez package on openwrt. 
I would like to use agent for setting the capability and authorizing the device. 
For example: "agent -c NoInputNoOutput 0000 &"
However, bluetoothctl is providing the NoInputNoOutput capability using the agent option but the pin as well is needed.

--
Thanks

jmv2009, 2018-01-10:
Have you tried btmgmt for the PIN. It may also work in bluetoothctl. Be sure you get "agent registered" message when you start bleutoothctl.

PraneetKaur, 2018-01-11:
Hi,
I was using agent in bluez 4.101 and it worked with "agent -c NoInputNoOutput 0000 &" command. Befire this, I enabled ssp mode using "hciconfig hci0 sspmode 1". This agent process runs in background and takes care of connecting automatically from my phone to the OpenWRT host without the need to enter the pin.
I need the same functionality in bluez5.37 and I found that bt-agent replaces the agent functionality in bluez 4.101. I am facing challenge in porting bt-agent to openwrt platform. Any help for the same would be appreciated.
Also I tried using "bluetoothctl -a NoInputNoOutput 0000 &" but couldnt really get it to work on OpenWRT.
Since I needed particularly 5.37 version of bluez as per the requirements, I am trying to get all needed functionalities working in bluez 5.37.
Attached are bluez4 and bluez5 agent and bt-agent respectively.

[image]
[image]

diizzyy, 2018-07-05:
@PraneetKaur 
Please close as there's no such package and/or 5.37 isn't supported (deprecated)

@hnyman @champtar

PraneetKaur, 2018-07-06:
Sure

--
Thanks
Praneet

closed 2018-07-06 by PraneetKaur