openwrt/packages #5424: python,python3: (discussion) how should source package dependencies be handled?

jefferyto, 2018-01-11:
Maintainer: @commodo 

Description:
(Not sure how many people to pull into this thread, @commodo feel free to mention others as necessary.)

Currently, only the main Python packages have source versions, where the source package depends on the source packages of its binary dependencies, i.e. if `python-A` depends on `libB` and `python-C`, then `python-A-src` depends on `python-C-src`.

I want to add more source packages, but I'm not sure if this way of structuring dependencies make sense. I can think of two ways to think about this (I'm sure there are more I haven't considered).

1. Under the current dependency pattern, source packages don't pull in any non-Python dependencies, so they can't be used without the binary package installed as well (unless the user manually installs any dependencies required). If this is the intention, that source packages aren't meant to be functional, then is it really necessary for source packages to have any dependencies?

    In the example above, installing `python-A-src` is not enough to use A because it doesn't pull in `libB`, so does `python-A-src` really need to depend on `python-C-src`?

2. If source packages are meant to be functional, then should they also have the same dependencies as their binary counterparts (i.e. should `python-A-src` also depend on `libB` and `python-C`)? Should they depend on their binary package directly (i.e. `python-A-src` depends on `python-A`)?

    I haven't fully investigated this, but are there other functionally necessary files in the binary package that doesn't exist in the source package (e.g. compiled C code, .egg-info directories, etc.)?

commodo, 2018-01-12:
hey,

Python source packages are a bit special, in the sense that they came up because I thought it would be nice to also package the Python source code, since the switch to packaging Python bytecodes by default.
[ I guess that was a bit obvious ]

I haven't given it much thought over how things should be done/designed.
I was hoping that there would be a bit more interest in them, from people complaining about them in one way or another.

But as it turns out, there does not seem to be much interest in this yet [aside from you].
People seem to be happy with Python bytecodes as default [for OpenWrt].

That means things can still be done in various ways moving forward.

I was also wondering if it makes sense to have src packages depend on each other.
For the Python interpreter packages, it's fine, but I think for other Python packages, it could be problematic, since some packages would choose to have `src` variants, and some wouldn't.

We could:
1. remove the DEPENDS scheme altogether
2. have it disabled by default, and add a flag to enable the dependencies

I am leaning towards 1. (right now), since 2. can offer potential for a mess.

Still, I don't have a strong preference ; if you have one, I'm open to hear it, and maybe we can apply that.

jefferyto, 2018-01-15:
For me, I don't mind packaging bytecodes (other major distros either package them as well or generate them post-install), but having source code available is important because:

1. It's easier to debug when source is available;
2. I like "free as in freedom" and being able to change any/every module anyway I want.

If the choice is between having no dependencies at all or having them disabled by default, I don't have a strong preference for either one; for my own uses I wouldn't go through the trouble of enabling dependencies. In terms of the above two reasons, I don't see the benefit of being forced to install the source code of all dependencies when I only want to debug or tinker with one specific module.

Removing dependencies sounds like less work so I would go for that (at least for now).

commodo, 2018-01-15:
Ok.
Then we disable all dependencies for -src packages :)

jefferyto, 2018-01-15:
Thanks :smile: closing for now.

closed 2018-01-15 by jefferyto