openwrt/packages #5445: more, more bug for build-scripts

build000, 2018-01-14:
```
date && make -j8 BUILD_LOG=1 && date
nie, 14 sty 2018, 21:11:30 CET
WARNING: Makefile 'package/feeds/routing/bird/Makefile' has a build dependency on 'libncurses', which does not exist
WARNING: Makefile 'package/feeds/routing/bird/Makefile' has a build dependency on 'libreadline', which does not exist
WARNING: Makefile 'package/feeds/packages/coova-chilli/Makefile' has a dependency on 'libmatrixssl', which does not exist
WARNING: Makefile 'package/feeds/telephony/freeswitch/Makefile' has a dependency on 'libsqlite2', which does not exist
WARNING: Makefile 'package/feeds/telephony/freeswitch/Makefile' has a dependency on 'flite', which does not exist
WARNING: Makefile 'package/feeds/telephony/freeswitch/Makefile' has a dependency on 'libyuv', which does not exist
WARNING: Makefile 'package/feeds/telephony/freeswitch/Makefile' has a dependency on 'hiredis', which does not exist
WARNING: Makefile 'package/feeds/telephony/freeswitch/Makefile' has a dependency on 'libilbc', which does not exist
WARNING: Makefile 'package/feeds/telephony/freeswitch/Makefile' has a dependency on 'libmemcached', which does not exist
WARNING: Makefile 'package/feeds/telephony/freeswitch/Makefile' has a dependency on 'libsilk', which does not exist
WARNING: Makefile 'package/feeds/telephony/freeswitch/Makefile' has a dependency on 'libg7221', which does not exist
WARNING: Makefile 'package/feeds/telephony/freeswitch/Makefile' has a dependency on 'freeradius-client', which does not exist
WARNING: Makefile 'package/feeds/luci/freifunk-p2pblock/Makefile' has a dependency on 'l7-protocols', which does not exist
WARNING: Makefile 'package/feeds/packages/libiio/Makefile' has a dependency on 'libavahi-common', which does not exist
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'kmod-usb-musb-tusb6010', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-diag-devinfo/Makefile' has a dependency on 'smap', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-diag-devinfo/Makefile' has a dependency on 'mac-to-devinfo', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-diag-devinfo/Makefile' has a dependency on 'smap-to-devinfo', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-diag-devinfo/Makefile' has a dependency on 'netdiscover-to-devinfo', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-pbx/Makefile' has a dependency on 'asterisk18', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-pbx/Makefile' has a dependency on 'asterisk18-app-authenticate', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-pbx/Makefile' has a dependency on 'asterisk18-app-disa', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-pbx/Makefile' has a dependency on 'asterisk18-app-setcallerid', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-pbx/Makefile' has a dependency on 'asterisk18-app-system', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-pbx/Makefile' has a dependency on 'asterisk18-chan-gtalk', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-pbx/Makefile' has a dependency on 'asterisk18-codec-a-mu', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-pbx/Makefile' has a dependency on 'asterisk18-codec-alaw', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-pbx/Makefile' has a dependency on 'asterisk18-func-cut', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-pbx/Makefile' has a dependency on 'asterisk18-res-clioriginate', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-pbx/Makefile' has a dependency on 'asterisk18-func-channel', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-pbx/Makefile' has a dependency on 'asterisk18-chan-local', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-pbx/Makefile' has a dependency on 'asterisk18-app-record', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-pbx/Makefile' has a dependency on 'asterisk18-app-senddtmf', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-pbx/Makefile' has a dependency on 'asterisk18-res-crypto', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-pbx-voicemail/Makefile' has a dependency on 'asterisk18', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-radvd/Makefile' has a dependency on 'radvd', which does not exist
WARNING: Makefile 'package/feeds/luci/luci-app-ushare/Makefile' has a dependency on 'ushare', which does not exist
WARNING: Makefile 'package/feeds/packages/python/Makefile' has a dependency on 'python-pip-conf-src', which does not exist
make[2]: Entering directory '/home/sj/Pulpit/Projekty/LEDE-REBOOT_FAST_PATH/WNDR4700_FAST_PATH/scripts/config'
make[2]: Leaving directory '/home/sj/Pulpit/Projekty/LEDE-REBOOT_FAST_PATH/WNDR4700_FAST_PATH/scripts/config'
```

barhom, 2018-01-26:
I see this as well;

$./scripts/feeds install -a
WARNING: Makefile 'package/feeds/telephony/freeswitch/Makefile' has a dependency on 'libsqlite2', which does not exist
WARNING: Makefile 'package/feeds/telephony/freeswitch/Makefile' has a dependency on 'flite', which does not exist
WARNING: Makefile 'package/feeds/telephony/freeswitch/Makefile' has a dependency on 'libyuv', which does not exist
WARNING: Makefile 'package/feeds/telephony/freeswitch/Makefile' has a dependency on 'hiredis', which does not exist
WARNING: Makefile 'package/feeds/telephony/freeswitch/Makefile' has a dependency on 'libilbc', which does not exist
WARNING: Makefile 'package/feeds/telephony/freeswitch/Makefile' has a dependency on 'libmemcached', which does not exist
WARNING: Makefile 'package/feeds/telephony/freeswitch/Makefile' has a dependency on 'libsilk', which does not exist
WARNING: Makefile 'package/feeds/telephony/freeswitch/Makefile' has a dependency on 'libg7221', which does not exist
WARNING: Makefile 'package/feeds/telephony/freeswitch/Makefile' has a dependency on 'freeradius-client', which does not exist

simdeveloper, 2018-01-28:
This command should do the trick as far as the freeswitch/Makefile problem goes:
./scripts/feeds uninstall freeswitch

hnyman, 2018-01-28:
I close this, as all other packages have already been fixed ages ago, and the maintainer of the freeswitch package is also working on the fix (see https://forum.openwrt.org/viewtopic.php?pid=371342#p371342 ).

In any case, the warnings are just warnings, but annoying :-(

closed 2018-01-28 by hnyman

omostan, 2018-01-28:
./scripts/feeds uninstall freeswitch works for me for the meantime.
Thanks for sharing!

build000, 2018-01-28:
Not working for me - still this other bug:
```
 make menuconfig
Collecting package info: done
WARNING: Makefile 'package/feeds/telephony/freeswitch/Makefile' has a dependency on 'libsqlite2', which does not exist
WARNING: Makefile 'package/feeds/telephony/freeswitch/Makefile' has a dependency on 'flite', which does not exist
WARNING: Makefile 'package/feeds/telephony/freeswitch/Makefile' has a dependency on 'libyuv', which does not exist
WARNING: Makefile 'package/feeds/telephony/freeswitch/Makefile' has a dependency on 'hiredis', which does not exist
WARNING: Makefile 'package/feeds/telephony/freeswitch/Makefile' has a dependency on 'libilbc', which does not exist
WARNING: Makefile 'package/feeds/telephony/freeswitch/Makefile' has a dependency on 'libmemcached', which does not exist
WARNING: Makefile 'package/feeds/telephony/freeswitch/Makefile' has a dependency on 'libsilk', which does not exist
WARNING: Makefile 'package/feeds/telephony/freeswitch/Makefile' has a dependency on 'libg7221', which does not exist
WARNING: Makefile 'package/feeds/telephony/freeswitch/Makefile' has a dependency on 'freeradius-client', which does not exist

*** End of the configuration.
*** Execute 'make' to start the build or try 'make help'.

```