openwrt/packages #5449: Snapshot Zerotier (1.2.4-2) package is broken.

Seohol, 2018-01-15:
- Device: TP-Link Archer C2600 v1.0 [**arm_cortex-a15_neon-vfpv4**]
- LEDE: LEDE Reboot 17.01.4 r3560-79f57e422d / LuCI lede-17.01 branch (git-17.290.79498-d3f0685)
- Steps to reproduce:

```
wget http://downloads.lede-project.org/snapshots/packages/arm_cortex-a15_neon-vfpv4/packages/zerotier_1.2.4-2_arm_cortex-a15_neon-vfpv4.ipk

opkg install zerotier_1.2.4-2_arm_cortex-a15_neon-vfpv4.ipk

Configuring zerotier.

Generate secret - please wait...
Segmentation fault
cat: can't open '/tmp/zt.sample_config.secret': No such file or directory
rm: can't remove '/tmp/zt.sample_config.secret': No such file or directory
```

zerotier cli also couldn’t start due to libstdcpp dependency.

[Report on reddit](https://www.reddit.com/r/zerotier/comments/74teow/unable_to_install_zerotier_package_on_lede_router/)

1.1.4 version from LEDE official repo works fine

@mwarning

mwarning, 2018-01-15:
The dependencies of the ZeroTier (ZT) package changed from ZT version 1.1.14 to 1.2.4. ZT package 1.2.2 started to depend on libstdcpp.

The problem is that the image/libstdcpp and ZT package are probably incompatible, since they are from different releases/snapshots.

Solutions:
1. Use a compatible image and ZT package combination, e.g.:
* http://downloads.lede-project.org/snapshots/targets/ipq806x/generic/openwrt-ipq806x-tplink_c2600-squashfs-sysupgrade.bin
* http://downloads.lede-project.org/snapshots/packages/arm_cortex-a15_neon-vfpv4/packages/zerotier_1.2.4-2_arm_cortex-a15_neon-vfpv4.ipk ('opkg update' and 'opkg install zerotier' should suffice)
2. install the libstdcpp package from snapshot as the ZT package. This might break other programs that depend on libstdcpp instead!

(Please don't be confused, LEDE and OpenWrt are more or less the same from a users perspective. LEDE is merging with OpenWrt as OpenWrt in the near future.)

mwarning, 2018-01-15:
The only real solution is to have a new OpenWrt release.
Updating the ZT package in the feed of the latest release might be an option, but this might not be accepted due to the package having changed dependencies..

mwarning, 2018-01-17:
@Seohol I'm going to close the ticket, as the problem is a imcompatible firmware/package version.
Please update the thread on reddit if you can.

mwarning, 2018-04-13:
The error has been found (ARM specific) and fixed:
https://github.com/openwrt/packages/pull/5908
:-)

dibdot, 2018-04-16:
Fixed with #5908

closed 2018-04-16 by dibdot