openwrt/packages #5458: collectd-mod-network: Consider enabling encryption support

u1f35c, 2018-01-16:
Maintainer: @hnyman 
Environment: BT Home Hub 5A / LEDE 17.04.1
Description:

The collectd-mod-network plugin is currently built without encryption support (`--without-libgcrypt` is explicitly passed to the configure script). It would be nice to have this support so that statistics can be securely transmitted off the router without having to use additional measures.

(This was previously raised at [https://dev.openwrt.org/ticket/18543](https://dev.openwrt.org/ticket/18543) but then closed with a pointer to GitHub.)

hnyman, 2018-01-16:
There was encryption possibility for a while, but it was removed some time ago as the build-time configuration proved really difficult (as apparently collectd upstream had faults in config handling)

u1f35c, 2018-01-17:
I downloaded the 17.01.4 SDK to see if there was something easy that could be fixed with collectd to allow this support, and discovered the CONFIG_PACKAGE_COLLECTD_ENCRYPTED_NETWORK option. Enabling this resulted in a collectd-mod-network package (5.5.3-4) with the appropriate support with no further work on my part - https://dev.openwrt.org/ticket/10604 suggests the issue was with an older collectd?

hnyman, 2018-01-17:
You are looking at really ancient bug discussions.
The encryption was enabled in 2016 and disabled in 2017 when 5.7.1 proved problematic. (it was impossible / rather difficult to completely remove crypto library linkage.

https://github.com/openwrt/packages/commits/master/utils/collectd

Discussion on feature removal:
https://github.com/openwrt/packages/pull/3917

u1f35c, 2018-01-17:
I *have* been looking at really ancient bug discussions, because those are what turned up when I tried to figure out why I couldn't talk to my Debian collectd server from my LEDE/OpenWRT router. Thanks for the pointers to the more recent discussions.

My understanding is that post 5.5.3 there were changes which caused compilation problems, an attempt was made to fix this but there were some issues about the libgcrypt dependency affecting more than collectd-mod-network and a lack of ability to test the changes so the support was just dropped entirely. Is this accurate? If so I can pull the current 5.8.0 package into my work tree and try to update and test things.

hnyman, 2018-01-17:
Yep, you have to understand that the buildbot will build the snapshots (and the releases) without encryption options, so it needs to compile cleanly without encryption. It seemed really difficult to prevent linking the crypt libraries even though the option was disabled in package build config. There was something wrong in collectd's build config logic.

I saw some recent commit in upstream collectd repo about fixing that, so it is possible that the linking problem has been fixed later, but so far nobody has invested the effort to really look into matters.

u1f35c, 2018-01-17:
Ok. Build with and without encryption tested, installation with both also tested. This seems to do the trick for me, but it does seem that the collectd gcrypt support in the configure scripts is somewhat fragile so it might break again with new upstreams. I can retest if/when that happens though.

closed 2018-01-21 by hnyman