openwrt/packages #5459: Transmission says port is closed even though it is open

felagund, 2018-01-16:
Maintainer: (not sure, mupliple people editing the makefile
Environment: Any recent LEDE (17.0.4). It does not matter which router is used.

Description: Even when one opens a port for transmission running on the router itself, Transmission still claims the port is closed: [image]

This is a longstanding issue as can be seen here: https://forum.lede-project.org/t/transmission-always-reports-listening-port-closed/2615/27

No matter what configuration we put into /etc/config/firewall, the port stays closed. There are no problems with running transmission on one's PC as opposed to the router itself.
```
config rule
        option target 'ACCEPT'
        option src 'wan'
        option proto 'tcp udp'
        option dest_port '51422'
        option name 'Transmission-router1'

config redirect
        option target 'DNAT'
        option src 'wan'
        option dest 'lan'
        option proto 'tcp udp'
        option src_dport '51422'
        option dest_ip '192.168.2.1'
        option dest_port '51422'
        option name 'trans?'

config rule
        option enabled '1'
        option target 'ACCEPT'
        option src 'wan'
        option dest 'lan'
        option name 'Transmission-router2'
        option dest_port '51422'
```

felagund, 2018-01-16:
Some other people having htis problem: https://forum.openwrt.org/viewtopic.php?id=62761
https://forum.openwrt.org/viewtopic.php?id=65856
https://forum.openwrt.org/viewtopic.php?id=70871
https://forum.openwrt.org/viewtopic.php?id=62250

It is reallt not a matter of misconfiguration.

felagund, 2018-01-16:
And one more: https://forum.lede-project.org/t/port-is-open-but-transmission-says-its-closed/2580/2

neheb, 2018-01-21:
It's probably an issue with the way transmission detects a closed port. It directs itself to visit a website and see the result. If anything fails, it reports a closed port. It should say unknown.

felagund, 2018-01-22:
What is weird is that it works when I install transmission on my PC and Transmission does not report an open port even when I try to use some scanner website and it says the port in question is open.

neheb, 2018-01-24:
Debugged this a little bit...

transmission seems to require ca-bundle to work properly. Also CURL_CA_BUNDLE needs to be set. Once all that is done, it manages to connect to the test webpage with an HTTP code of 200.

This might need adjustments to curl. Will investigate.

felagund, 2018-01-24:
Thanks for investigation!

neheb, 2018-01-30:
Good news. I found the issue. The libcurl in OpenWrt is built by default with zlib and verbose error strings disabled. After enabling both, The port checking feature correctly shows open. I'll figure out which but in the meantime, just ignore what transmission tells you.

vasilis74, 2018-02-01:
@neheb If you really found the issue, you will please many people very much! I suppose the fix can be applied to LEDE, right?
Could you help us enable zlib and verbose error strings, while you search for an exact solution? If yes, please give newbie directions, me at least need a copy-paste guide to configure files and services. And please specify how your fix applies to LEDE if you can, because some configuration files are not the same as OpenWRT.
Sorry if I ask for too much and many thanks for any help!

neheb, 2018-02-01:
@vasilis74 pull request for transmission is here: https://github.com/transmission/transmission/pull/495

Apply the patch to your local build. Or wait until I submit my version bump in 1 or 2 days. I'm testing out some stuff currently.

vasilis74, 2018-02-01:
Thanks again! I will wait for a method I can follow.

neheb, 2018-02-03:
Latest version fixes the issue. Please close.

felagund, 2018-02-04:
Thanks, taht was quick!

closed 2018-02-04 by felagund