openwrt/packages #5461: node: update to the latest LTS release (v8.9.4)

artynet, 2018-01-16:
Hello @NeoRaider @ianchi @nxhack,

I think the time to upgrade node.js to its latest LTS version (v8.9.4) has finally arrived. The v6 series is now considered outdated by the very node.js community itself....@nxhack might take care of the this since his work is basically ready. Please let me know your thoughts about that.

Best, Arturo

nxhack, 2018-01-17:
Hi @artynet 

On the official node.js website v8.x is described as "Recommended For Most Users".

My custom package can also build v 8.9.4, and it also supports module package creation with npm 5.6.0. It also delete a lot of cache and tmp file which npm use. (It causes problems on buildbots.)

In fact, I am still using node.js v6.x. (AR9331, extroot, swap)

I am worried that the size of v8.x is large.
```
node_v6.12.3-1_mips_24kc.ipk (3,600,425 bytes)
node-npm_v6.12.3-1_mips_24kc.ipk (2,728,855 bytes)

node_v8.9.4-1_mips_24kc.ipk (4,591,508 bytes)
node-npm_v8.9.4-1_mips_24kc.ipk (5,250,658 bytes)
```
However, I think that should be updated to v8.x.

diizzyy, 2018-07-05:
@artynet 
I think this can be closed as of https://github.com/openwrt/packages/commit/9ce4d5191f97f2463fb6426fd6bc099a35dbf662 ?

closed 2018-07-05 by yousong