openwrt/packages #5464: Transmission: better to update to latest git to address CVE-2018-5702

DazzyWalkman, 2018-01-17:
Recently it was found that transmission-daemon was vulnerable to dns rebinding attacks.
 [https://vuxml.freebsd.org/freebsd/3e5b8bd3-0c32-452f-a60e-beab7b762351.html](url)
and transmission was updated to address this issue.
https://github.com/transmission/transmission/commit/eb5d1a79cbe1b9bc5b22fdcc598694ecd4d02f43

The default local DNS server (dnsmasq) of OpenWRT/LEDE already supports DNS Rebinding Protections.  I noticed in syslog it had prevented numerous such attacks against transmission-daemon.

Still it's not a bad thing to update related transmission packages as well.

neheb, 2018-01-21:
The change breaks doing something like http://lede:9091 to access transmission. I don't think this is appropriate. Will investigate a solution.

neheb, 2018-01-28:
I've updated to 2.93 and disabled it by default as it's not needed by OpenWrt. Please close.

closed 2018-01-28 by DazzyWalkman