openwrt/packages #5467: radicale: upgrade to 2.x

DragoonAethis, 2018-01-17:
Maintainer: @chris5560 
Environment: LEDE 17.01.4

Radicale 2.x (currently at 2.1.8) was released a few months ago. Now that it had some time to mature, it'd be a solid idea to upgrade the package in this repo. The 2.x branch now requires Python 3.x (dropping 2.x completely), but consumes a bit less memory than 1.x and (*finally*) properly synchronizes calendars which use a bit more obscure iCal features (date-based filters in recurring events, per-event colors, etc).

2.x isn't backwards compatible with 1.x data - storage needs to be migrated first, [as described on this page](http://radicale.org/1to2/). This can be automated, but then Radicale still needs to be reconfigured. It'd be preferred to leave 1.x up and running in LEDE/OpenWrt 17.01.x and upgrade to 2.x in the next release.

chris5560, 2018-01-17:
Hi Dragoon,

please feel free to build a 2.x package and take over "Maintainer" for it.
I'm currently short on time to build and test a 2.x package.
I think we should have two packages for a while 1.x and 2.x, so user can decide when to change/upgrade independent from OpenWrt release.
I offer my support to (re)write a corresponding Luci application.

Christian

DragoonAethis, 2018-01-17:
I never did any packages for OpenWrt previously, but I'll try to package it in about 2 weeks.

jefferyto, 2019-01-08:
@DragoonAethis you may be interested in #7871.

DragoonAethis, 2019-01-13:
Awesome, thank you! I'll try to test it soon :)

closed 2019-01-13 by DragoonAethis