openwrt/packages #5474: openconnect: missing/does not build

adagari, 2018-01-18:
I was trying to install openconnect on my router. The packages is missing and upon further investigation I learned the packages is failing to build. Here is the log from the server
https://downloads.lede-project.org/snapshots/faillogs/arm_cortex-a9_vfpv3/packages/openconnect/compile.txt

hnyman, 2018-01-20:
maintainer: @nmav 

in buildbot, openconnect fails to build for some target platforms (e.g. mips_24kc, arm_cortex-a9_vfpv3) and compiles for others (e.g. arm_cortex-a15_neon-vfpv4).

The reason seems to be a dependency error related to krb5, but I haven't tried to find out the deeper reason why some targets compile ok.
```
Package openconnect is missing dependencies for the following libraries:
libgssapi_krb5.so.2
```
Most likely at some targets the package's configure script finds the krb5 package and tries to link for it, although it is not needed?

Could be something similar as with 
https://github.com/openwrt/packages/commit/fa4365dd9b50a5bc032498d6f281daeefe02e49e
or
https://github.com/openwrt/packages/commit/4c210398846da42b010345155f5cc0e1bd5ed677

nmav, 2018-01-20:
Thanks. I've  disabled any libraries we don't explicitly require in openwrt.

nmav, 2018-01-20:
Are the buildbot results publicly available?

closed 2018-01-20 by nmav

hnyman, 2018-01-21:
Yes in the faillogs for phase2
https://downloads.lede-project.org/snapshots/faillogs/

So far only a few targets have been rebuilt after the change, so most of the logs still reflect older builds
https://phase2.builds.lede-project.org/grid

commit aa93ca86 (2018-01-20): openconnect: build without options we don't support

That is, since we don't require gssapi or libpskc, avoid
accidental builds with it.

Closes #5474