openwrt/packages #5486: extending OpenWrt packages maintainers by a new member

lperkov, 2018-01-22:
Dear all,

I would like to propose granting Marko Ratkaj (ratkaj on GitHub) commit access.

Over the last few years that I know him personally, Marko has done a lot of great work. He maintains a dozen of packages and is actively porting new ones to OpenWrt. He is very knowledgeable about
OpenWrt/LEDE in general and he actively pushes new packages and fixes issues in existing ones within the packages feed.

That being said, he has my vote to gain access to the above mentioned feed.

Thanks,
Luka

@karlp @tripolar @pprindeville @Noltari @jmccrohan @tohojo @NeoRaider @rmilecki @nmeyerhans @kerneis @zioproto @luizluca @acinonyx @stintel @mhei @nmav @heil @nunojpg @thess @champtar @yousong @jslachta @hnyman @Naoir @dangowrt @ratkaj

hnyman, 2018-01-22:
Why not :-)

thess, 2018-01-22:
ACK from me.

/ted

From: Luka Perkov
Sent: Monday, January 22, 2018 9:22 AM
To: openwrt/packages
Cc: Ted Hess ; Mention
Subject: [openwrt/packages] extending OpenWrt packages maintainers by a new member (#5486)

Dear all,

I would like to propose granting Marko Ratkaj (ratkaj on GitHub) commit access.

Over the last few years that I know him personally, Marko has done a lot of great work. He maintains a dozen of packages and is 
actively porting new ones to OpenWrt. He is very knowledgeable about
OpenWrt/LEDE in general and he actively pushes new packages and fixes issues in existing ones within the packages feed.

That being said, he has my vote to gain access to the above mentioned feed.

Thanks,
Luka

@karlp @tripolar @pprindeville @Noltari @jmccrohan @tohojo @NeoRaider @rmilecki @nmeyerhans @kerneis @zioproto @luizluca @acinonyx 
@stintel @mhei @nmav @heil @nunojpg @thess @champtar @yousong @jslachta @hnyman @Naoir @dangowrt @ratkaj

thess, 2018-01-22:
ACK for Dirk also.

/ted

From: Hannu Nyman 
Sent: Monday, January 22, 2018 11:52 AM
To: openwrt/packages 
Cc: Ted Hess ; Mention 
Subject: Re: [openwrt/packages] extending OpenWrt packages maintainers by a new member (#5486)

On the same track, I propose also Dirk Brenken (@dibdot ), who is authoring several popular packages (e.g. adblock, travelmate) and maintaining them at regular intervals.

pprindeville, 2018-01-22:
No objection to either.

champtar, 2018-01-23:
Fine with me, the more the merrier!

nmav, 2018-01-23:
Fine with me too. Welcome!

jslachta, 2018-01-23:
Full ACK.

stintel, 2018-01-23:
ACK for both

zioproto, 2018-01-23:
> ACK from me as well

Hello,

I dont know the new guys but for me it is always good to be inclusive and
welcome new people that to work for OpenWrt.
Welcome in ! ACK from me.

thess, 2018-01-23:
Invites sent to both Marko and Dirk. Closing this for now.
Nice to have to new folks here...

closed 2018-01-23 by thess

ratkaj, 2018-01-24:
Thank you everyone for your support!

dibdot, 2018-01-24:
Many thanks for your confidence and your long lasting support. I am looking forward to working with  you ...

Best regards
Dirk