openwrt/packages #5487: Pulseaudio + Bluez + A2dp stereo + Openwrt

jmv2009, 2018-01-22:
Pulseaudio + Bluez + A2dp stereo + Openwrt

You have to rebuild openWRT.
Remove --disable-dbus and add +sbc +libdbus after e.g. +alsa-lib in feeds/packages/sound/pulseaudio/Makefile

In make menuconfig make sure at least pulseaudio-daemon and bluez-daemon are going to be available (e.g. built in)
Do this after getting the source and feeds with git. The bluetooth pulseaudio modules should build (in the 
source/build_dir/target-XXX/pulseaudio-11.1/src folder)
Furthermore, once on the OpenWRT system use

<busconfig>
    
    <policy context="default">
        <allow own="org.pulseaudio.Server"/>
	<allow send_type="method_call"/>
	<allow send_type="method_return"/>
    </policy>

 </busconfig>
for pulseaudio.conf in /etc/dbus-1/system.d

and

<busconfig>

  

  <policy user="root">
    <allow own="org.bluez"/>
    <allow send_type="method_call"/>
    <allow send_type="method_return"/>
  </policy>

  <policy at_console="true">
    <allow send_path="/"/>
    <allow send_path="/org/bluez"/>

    <allow send_destination="org.bluez.Manager"/>
    <allow receive_sender="org.bluez.Manager"/>

    <allow send_destination="org.bluez.Adapter"/>
    <allow receive_sender="org.bluez.Adapter"/>

    <allow send_destination="org.bluez.Device"/>
    <allow receive_sender="org.bluez.Device"/>

    <allow send_destination="org.bluez.Service"/>
    <allow receive_sender="org.bluez.Service"/>

    <allow send_destination="org.bluez.Database"/>
    <allow receive_sender="org.bluez.Database"/>

    <allow send_destination="org.bluez.Security"/>
    <allow receive_sender="org.bluez.Security"/>
  </policy>

</busconfig>
for bluetooth.conf in /etc/dbus-1/system.d

add

load-module module-bluez5-discover
load-module module-bluetooth-policy
to /etc/pulse/system.pa

remove
--disable-module-loading
from
/etc/init.d/pulseaudio

For avahi/zeroconf,
remove --disable-avahi , and add +libavahi-client to the DEPENDS in the pulseaudio Makefile
and add load-module module-zeroconf-publish to /etc/pulse/system.pa on the openwrt system.

After modification of /etc/pulse/system.pa issue service pulseaudio restart

It's not working as well as on kali linux. Both run Bluez 5.47 and pulseaudio 11.1. But Openwrt only works as a a2dp sink when connecting from windows upon pairing.

Run bluetoothctl, and issue

power on
discoverable on
connect from windows
trust XX:XX:XX:XX:XX:XX
disconnect from windows
connect from windows
once you get "Connected music" you're golden. You can set the discoverable time in /etc/bluetooth/main.conf

Things may depend a bit on the bluetooth version for the pairing. In my case, when doing it this way no pin or other authentication was required. Note: pulseaudio/bluez do not suspend the stream when there is no audio. So if you have had some loss of packets, the audio stream may get interrupted and delayed, and you pretty much need to remove the device to resync.

I'm trying to figure out why I don't get Connected Music when reconnecting without repairing. The output from bluetoothd -n -d after service bluetoothd stop should give this information, but I haven't deciphered it. It now looks the main problem is with disconnection. Connected Music also works when connecting from Openwrt when it's already paired.

Further open issues: Can somebody verify the dbus files?
Can this be integrated into standard openwrt?
There are a couple of issues:
A) Verification
B) how to avoid incorporating dbus if you don't need it.
C) How to deal with the --disable-module-loading requirement, as it typically is recommended against.

jmv2009, 2018-01-23:
I think if the commit at
https://github.com/openwrt/packages/commit/4b1fad86f85bed0d867324c21d7e4a75d9957bfc is reversed, bluetooth would work.

jmv2009, 2018-01-23:
or use 
```
<busconfig>

  <!-- System-wide PulseAudio runs as 'pulse' user. This fragment is
       not necessary for user PulseAudio instances. -->

  <policy user="pulse">
    <allow own="org.pulseaudio.Server"/>
  </policy>

</busconfig>
```
for pulsaudio-system.conf from the pulseaudio source??

jmv2009, 2018-01-25:
This Makefile simultaneously enables avahi/zeroconf and bluez/a2dp when the package pulseaudio-daemon-avahi is activated: It is a reversal of 4b1fad8, with the sbc library dependency added. Pulseaudio-daemon is not affected, and is a parallel option.

```
#
# Copyright (C) 2011-2016 OpenWrt.org
#
# This is free software, licensed under the GNU General Public License v2.
# See /LICENSE for more information.
#

include $(TOPDIR)/rules.mk

PKG_NAME:=pulseaudio
PKG_VERSION:=11.1
PKG_RELEASE:=1

PKG_SOURCE:=$(PKG_NAME)-$(PKG_VERSION).tar.xz
PKG_SOURCE_URL:=https://freedesktop.org/software/pulseaudio/releases/
PKG_HASH:=f2521c525a77166189e3cb9169f75c2ee2b82fa3fcf9476024fbc2c3a6c9cd9e
PKG_LICENSE:=LGPL-2.1+
PKG_LICENSE_FILES:=GPL LICENSE

PKG_BUILD_DEPENDS:=intltool/host

PKG_FIXUP:=autoreconf
PKG_USE_MIPS16:=0

PKG_INSTALL = 1

include $(INCLUDE_DIR)/package.mk
include $(INCLUDE_DIR)/nls.mk

define Package/pulseaudio/Default
  SECTION:=sound
  CATEGORY:=Sound
  DEPENDS:=+libsndfile +libltdl +libpthread +librt +alsa-lib \
	+libopenssl +libwrap +libcap $(ICONV_DEPENDS) $(INTL_DEPENDS)
  TITLE:=Network sound server
  MAINTAINER:=Peter Wagner <tripolar@gmx.at>
  URL:=http://www.pulseaudio.org
  PROVIDES:=pulseaudio
  USERID:=pulse=51:pulse=51
endef

define Package/pulseaudio-daemon
  $(call Package/pulseaudio/Default)
  VARIANT:=noavahi
endef

define Package/pulseaudio-daemon-avahi
  $(call Package/pulseaudio/Default)
  DEPENDS+=+dbus +libavahi-client +avahi-daemon +sbc
  TITLE+= (avahi/bluez)
  VARIANT:=avahi
endef

define Package/pulseaudio/Default/description
  PulseAudio (formerly Polypaudio) is a cross-platform, networked sound server.
endef

define Package/pulseaudio-daemon/description
  $(call Package/pulseaudio/Default/description)
endef

define Package/pulseaudio-daemon-avahi/description
  $(call Package/pulseaudio/Default/description)
  This package enables avahi,bluez and is compiled against dbus, sbc, and avahi.
endef

define Package/pulseaudio-daemon/conffiles
[... 181 lines cut ...]

	$(INSTALL_BIN) \
		$(PKG_INSTALL_DIR)/usr/bin/pa* \
		$(1)/usr/bin/
endef

define Package/pulseaudio-profiles/install
	$(INSTALL_DIR) \
		$(1)/usr/share/pulseaudio/alsa-mixer/paths \
		$(1)/usr/share/pulseaudio/alsa-mixer/profile-sets

	$(INSTALL_DATA) \
		$(PKG_INSTALL_DIR)/usr/share/pulseaudio/alsa-mixer/paths/* \
		$(1)/usr/share/pulseaudio/alsa-mixer/paths

	$(INSTALL_DATA) \
		$(PKG_INSTALL_DIR)/usr/share/pulseaudio/alsa-mixer/profile-sets/* \
		$(1)/usr/share/pulseaudio/alsa-mixer/profile-sets
endef

$(eval $(call BuildPackage,pulseaudio-daemon))
$(eval $(call BuildPackage,pulseaudio-daemon-avahi))
$(eval $(call BuildPackage,pulseaudio-tools))
$(eval $(call BuildPackage,pulseaudio-profiles))
```

diizzyy, 2018-05-05:
https://github.com/openwrt/packages/pull/6010 might be of interest

jmv2009, 2018-05-06:
Superseded, #5577

closed 2018-05-06 by jmv2009