openwrt/packages #5496: Transmission unstable: SIGSEGV to transmission-da for invalid write

lucasfarpadus, 2018-01-24:
transmission, Transmission-daemon

Maintainer: @neheb 
Environment: SoC: Atheros AR9330 rev 1; TL-WR710N v1; LuCI lede-17.01 branch (git-17.290.79498-d3f0685) / LEDE Reboot 17.01.4 r3560-79f57e422d

Description:
The torrent download stop and then continues. When download finally end, the error stops too.
Transmission Web Interface stop responding while the error occurs.

Kernel Log
(...)
[   29.837100] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   31.104026] br-lan: port 1(eth1) entered forwarding state
(... until here, everything is ok)
(... then I start the torrent download)
[21631.332044] 
[21631.332044] do_page_fault(): sending SIGSEGV to transmission-da for invalid write access to 00000000
[21631.339977] epc = 77485ca8 in libc.so[77414000+92000]
[21631.344756] ra  = 773e3393 in libevent-2.0.so.5.1.10[773d6000+3b000]
[21631.351108] 
[22180.002056] 
[22180.002056] do_page_fault(): sending SIGSEGV to transmission-da for invalid write access to 00000000
[22180.009977] epc = 77a89ca8 in libc.so[77a18000+92000]
[22180.014775] ra  = 779e7393 in libevent-2.0.so.5.1.10[779da000+3b000]
[22180.021162] 
[22227.819743] 
[22227.819743] do_page_fault(): sending SIGSEGV to transmission-da for invalid write access to 00000000
[22227.827687] epc = 7739dcdc in libc.so[7732c000+92000]
[22227.832464] ra  = 772fb393 in libevent-2.0.so.5.1.10[772ee000+3b000]
[22227.838826] 
[22280.792186] 
[22280.792186] do_page_fault(): sending SIGSEGV to transmission-da for invalid write access to 00000000
[22280.799971] epc = 77ef7ca8 in libc.so[77e86000+92000]
[22280.804905] ra  = 77e55393 in libevent-2.0.so.5.1.10[77e48000+3b000]
[22280.811235] 
[22344.663198] 
[22344.663198] do_page_fault(): sending SIGSEGV to transmission-da for invalid write access to 00000000
[22344.671220] epc = 77af7ca8 in libc.so[77a86000+92000]
[22344.676114] ra  = 77a55393 in libevent-2.0.so.5.1.10[77a48000+3b000]
[22344.682247] 
[22394.894085] 
[22394.894085] do_page_fault(): sending SIGSEGV to transmission-da for invalid write access to 00000000
[22394.902078] epc = 77af9cdc in libc.so[77a88000+92000]
[22394.906802] ra  = 77a57393 in libevent-2.0.so.5.1.10[77a4a000+3b000]
[22394.913229] 
[22451.933165] 
[22451.933165] do_page_fault(): sending SIGSEGV to transmission-da for invalid write access to 00000000
[22451.940963] epc = 77b7dca8 in libc.so[77b0c000+92000]
[22451.945884] ra  = 77adb393 in libevent-2.0.so.5.1.10[77ace000+3b000]
[22451.952213] 
[22493.793730] 
[22493.793730] do_page_fault(): sending SIGSEGV to transmission-da for invalid write access to 00000000
[22493.801514] epc = 77a67ca8 in libc.so[779f6000+92000]
[22493.806469] ra  = 779c5393 in libevent-2.0.so.5.1.10[779b8000+3b000]
[22493.812775] 
[22545.840368] 
[22545.840368] do_page_fault(): sending SIGSEGV to transmission-da for invalid write access to 00000000
[22545.848223] epc = 772e3ca8 in libc.so[77272000+92000]
[22545.853217] ra  = 77241393 in libevent-2.0.so.5.1.10[77234000+3b000]
[22545.859503] 
[22668.661662] 
[22668.661662] do_page_fault(): sending SIGSEGV to transmission-da for invalid write access to 00000000
[22668.669618] epc = 7749bca8 in libc.so[7742a000+92000]
[22668.674561] ra  = 773f9393 in libevent-2.0.so.5.1.10[773ec000+3b000]
[22668.680888]

neheb, 2018-01-25:
Seems like a big endian bug. I'm not running any big endian systems currently so I can't really test.

Can you try running "transmission-daemon --log-debug -f"? Hopefully it gives some hints.

lucasfarpadus, 2018-01-26:
1 - I run the command "transmission-daemon --log-debug -f" over ssh as root and no problem reported, but download not work because the missing of config file.
2 - I run the command "/usr/bin/transmission-daemon --log-debug --logfile /mnt/sda3/torrents/log.log -g /etc/transmission -f" over ssh as root, no problem reported and the download works fine.
3 - Not convinced, I put the "--log-debug --logfile /mnt/sda3/torrents/log.log" inside the file "/etc/init.d/transmission" (Item CMDLINE, in my case, line 110). Result: The error again. I attach the logs here. 
Runing as root or common user result the same error.

Without the extra commands that you send me, the log say:
epc = 7749bca8 in libc.so[7742a000+92000]
ra = 773f9393 in libevent-2.0.so.5.1.10[773ec000+3b000]

With the extra commands
 epc = 00425805 in transmission-daemon[400000+47000]
ra  = 004257d7 in transmission-daemon[400000+47000]

[log_from_daemon.log](https://github.com/openwrt/packages/files/1666531/log_from_daemon.log)
[log_from_kernel.log](https://github.com/openwrt/packages/files/1666532/log_from_kernel.log)
[log_from_system.log](https://github.com/openwrt/packages/files/1666533/log_from_system.log)

.

neheb, 2018-01-26:
That's weird that transmission when ran manually works but doesn't work
with the init script. The one difference I know of is that by default,
transmission runs as it's own user and not root.

One other one is that the init script has some support for procd jails.
That may also be playing a factor. Will investigate.

dangowrt, 2018-01-26:
The problem here is that procd's seccomp jail conflicts with setting
the user parameter -- because the seccomp jail is applied before the
user is changed, which then fails. Doing it the other way arround also
can't easily work, as doing the chroot call only workds as root.

It will have to be an interleaved process of setting process parameters
and dropping the priviledges needed for each step after it is
performed.

neheb, 2018-01-28:
@dangowrt procd_add_jail is called after procd_set_param user. If the jail is actually the problem, I'll probably end up getting rid of it.

@lucasfarpadus can you modify the init script in /etc/init.d/transmission and remove all of the procd_add_jail settings to see if it fixes the issue?

In other news I am about to push an update to 2.93 which just got released. Also try that.

lucasfarpadus, 2018-01-30:
Results:
The download begins and then...

Mon Jan 29 23:27:50 2018 kern.info kernel: [  117.630625] 
Mon Jan 29 23:27:50 2018 kern.info kernel: [  117.630625] do_page_fault(): sending SIGSEGV to transmission-da for invalid write access to 00000000
Mon Jan 29 23:27:50 2018 kern.info kernel: [  117.638642] epc = 77df7ca8 in libc.so[77d86000+92000]
Mon Jan 29 23:27:50 2018 kern.info kernel: [  117.643406] ra  = 77d55393 in libevent-2.0.so.5.1.10[77d48000+3b000]
Mon Jan 29 23:27:50 2018 kern.info kernel: [  117.649671] 

...The download stop.

I tried to add a comment and delete the lines, but no effect. 
#	procd_add_jail transmission log
#	procd_add_jail_mount $config_file
#	procd_add_jail_mount_rw $download_dir
	procd_close_instance

lucasfarpadus, 2018-01-30:
I done a fresh install of the router with LEDE and transmission and same problem occur. Is just me? I'm think its my SOC the real problem.

neheb, 2018-01-30:
The weird thing is that you said it works fine if ran manually. This implies the init script is at fault.

The only other thing I can think of is the memory limit that procd places on transmission.

Actually that's probably it. You're using transmission on a 32MB router limited to 16MB. Some memory allocation is probably failing.

Try deleting or commenting the lines above the jail one starting with if test

neheb, 2018-02-02:
To fix this, edit /etc/config/transmission and change

option mem_percentage '50'

to

option mem_percentage '100'

I'm seriously considering removing this option.

lucasfarpadus, 2018-02-06:
> option mem_percentage '100'
Really fixed.

Now I'm trying to understand why my download are too slow.
Appears to be occurs only when I use the internet beyond the torrents in the router. There is a priority option to set?

Btw. I think its another topic, or could you give me a way?

neheb, 2018-02-07:
I don't think you can expect much from a 4/32MB router. Those things are so low end.

BTW are you saying a previous version of transmission is faster?

lucasfarpadus, 2018-02-07:
> BTW are you saying a previous version of transmission is faster?

No. It is the first time that I use a router as file storage/downloader and Transmission on Linux.

So, I just want to say thank you for your support.

closed 2018-02-07 by lucasfarpadus