openwrt/packages #5498: fwknop: OPEN_PORTS is ignored when a client puts open ports in the SPA

waitfor1t, 2018-01-25:
Maintainer: @jp-bennett 
Environment: all/all

Description:
fwknop: OPEN_PORTS is ignored when a client adds open ports in the SPA. This is a security risk; it allows any client to open any port. The OPEN_PORTS variable, if specified, should ignore whatever the client sends.

jp-bennett, 2018-01-25:
On the latest LEDE release, I can't replicate this problem.  Can you sanitize and share the log file where this happened, and also check whether /tmp/etc/access.conf does indeed include the OPEN_PORTS line correctly?

waitfor1t, 2018-01-25:
/tmp/etc/access.conf does have the OPEN_PORTS line. I've tried with spaces and without (opening multiple ports). Now I'll try a few combinations (1 port, 2 without space, etc) to see if that changes anything.

jp-bennett, 2018-01-25:
@waitfor1t You might find the problem disappears with the latest release, but keep me updated.  If it's still an issue I want to get it fixed.

waitfor1t, 2018-01-27:
so I'm now running the latest LEDE build and can see that the server (fwknopd 2.6.9-3) does reject any SPA from the client that doesn't match the server, so it seems resolved. closing the issue.

closed 2018-01-27 by waitfor1t