openwrt/packages #5512: fftw3: hangs on raspberry pi3

soundart, 2018-01-26:
-------------------------------

Maintainer: @Wedmer 
Environment: (arm, brcm2708_bcm2710, lede 79f57e422d229bb21837f588f40259835893837b)

Description:

FFTW3 hangs due to using CLOCK_SGI_CYCLE, which was removed from linux.
see https://github.com/FFTW/fftw3/issues/122 and https://github.com/FFTW/fftw3/pull/123

```
diff --git a/kernel/cycle.h b/kernel/cycle.h
index fe3dd50d..777d1cda 100644
--- a/kernel/cycle.h
+++ b/kernel/cycle.h
@@ -437,14 +437,24 @@ INLINE_ELAPSED(__inline)
 #define HAVE_TICK_COUNTER
 #endif
 /*----------------------------------------------------------------*/
-/* SGI/Irix */
-#if defined(HAVE_CLOCK_GETTIME) && defined(CLOCK_SGI_CYCLE) && !defined(HAVE_TICK_COUNTER) && !defined(__ANDROID__)
+/* SGI/Irix/Linux but not android */
+#if !defined(__ANDROID__) && !defined(HAVE_TICK_COUNTER)
+#if defined(HAVE_CLOCK_GETTIME)
+#if defined(CLOCK_MONOTONIC)
+#define METHOD CLOCK_MONOTONIC
+#elif defined(CLOCK_REALTIME)
+#define METHOD CLOCK_REALTIME
+#elif defined(CLOCK_SGI_CYCLE)
+#define METHOD CLOCK_SGI_CYCLE
+#endif
+#endif
+
 typedef struct timespec ticks;

 static inline ticks getticks(void)
 {
      struct timespec t;
-     clock_gettime(CLOCK_SGI_CYCLE, &t);
+     clock_gettime(METHOD, &t);
      return t;
 }

```

Wedmer, 2018-01-26:
I think we should wait for next fftw3 release

diizzyy, 2018-07-05:
@soundart 
fftw3 has been bumped, is it still broken?
If not, please close this issue report.

soundart, 2018-07-05:
I cannot find the fftw3 3.3.8 tag on github currently, but cycle.h on master ist unchanged since november last year and my pull request is still open.

squircle, 2020-07-06:
fftw3 was removed by e0777387514c02201c8f38a6ed47887103a276a8, so this issue can be closed.

soundart, 2020-07-06:
I have used it in an openwrt context as part of a media player.

closed 2020-07-06 by soundart