openwrt/packages #5525: error build/compile transmission 2.93 (patching error 040-fix-for-mbedtls.patch)

build000, 2018-01-28:
version OpenWrt: 
```
scripts/getver.sh
r5972-74beb6f
```
make -j8 BUILD_LOG=1
cat logs/package/feeds/packages/transmission/openaal/compile.txt:
```
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://github.com/transmission/transmission-releases/raw/master/transmission-2.93.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed

  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100   166  100   166    0     0    164      0  0:00:01  0:00:01 --:--:--   165

  0     0    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0
 74 3285k   74 2447k    0     0  1168k      0  0:00:02  0:00:02 --:--:-- 2538k
100 3285k  100 3285k    0     0  1492k      0  0:00:02  0:00:02 --:--:-- 3070k

Applying ./patches/020-use-internal-miniupnp.patch using plaintext: 
patching file configure.ac
Hunk #1 succeeded at 378 (offset -8 lines).

Applying ./patches/040-fix-for-mbedtls.patch using plaintext: 
patching file configure.ac
Hunk #1 FAILED at 154.
1 out of 1 hunk FAILED -- saving rejects to file configure.ac.rej
Patch failed!  Please fix ./patches/040-fix-for-mbedtls.patch!
Makefile:199: recipe for target '/home/sj/Pulpit/Projekty/WNDR4700-TRUNK/build_dir/target-powerpc_464fp_musl/transmission-openssl/transmission-2.93/.prepared_edb44707a9d7ea7e2dd2b8deb549f04e_6664517399ebbbc92a37c5bb081b5c53' failed
make[3]: *** [/home/sj/Pulpit/Projekty/WNDR4700-TRUNK/build_dir/target-powerpc_464fp_musl/transmission-openssl/transmission-2.93/.prepared_edb44707a9d7ea7e2dd2b8deb549f04e_6664517399ebbbc92a37c5bb081b5c53] Error 1
```

obsy, 2018-01-28:
@neheb is the maintainer

neheb, 2018-01-28:
Will fix shortly

build000, 2018-01-28:
sorki Cezary - nie spojrzałem w Makefile - nie wiedziałem, że już nie jesteś opiekunem pakietu.

build000, 2018-01-28:
https://github.com/openwrt/packages/commit/3b3fed7fb8a3274011897f048a8edb38797b2735 - this patch fixed problem...

...of course, when a patch is used, it has not been added to official sources for unknown reasons, namely:
eq.
061-disable-po-directory.patch
```
--- a/configure.ac
+++ b/configure.ac
@@ -660,8 +660,7 @@
                  web/style/transmission/images/Makefile
                  web/style/transmission/images/buttons/Makefile
                  web/javascript/Makefile
-                 web/javascript/jquery/Makefile
-                 po/Makefile.in])
+                 web/javascript/jquery/Makefile])
 
 AC_OUTPUT
 
--- a/Makefile.am
+++ b/Makefile.am
@@ -44,7 +44,6 @@
   gtk/CMakeLists.txt \
   libtransmission/CMakeLists.txt \
   libtransmission/version.h.in \
-  po/CMakeLists.txt \
   qt/CMakeLists.txt \
   utils/CMakeLists.txt
 ```

neheb, 2018-01-28:
New version here: https://github.com/openwrt/packages/pull/5526

Local builds compile fine, travis does as well. Sorry for the fuss.

closed 2018-01-28 by build000

build000, 2018-01-30:
Confirm this patch/commit: https://github.com/openwrt/packages/commit/b3710f578afab4848decee2ce7d8543942706cc8
- build/compile properly.