openwrt/packages #5528: p11-kit: compile fails, no trust module paths

hewonen, 2018-01-29:
Maintainer: @\<n.mavrogiannopoulos@gmail.com>
Environment: mips_24kc, TP-Link Archer C7, LEDE v17.01.4

Description:

I have selected ca-bundle and ca-certificates as modules, if this has anything to do with it.

```
checking if trust module is enabled... yes
checking for trust module paths... configure: error: could not find. Use --with-trust-paths=/path to set, or --without-trust-paths to disable
make[3]: *** [/Volumes/ledes/c7source/build_dir/target-mips_24kc_musl-1.1.16/p11-kit-0.23.1/.configured_yyyyyyy] Error 1
make[3]: Leaving directory `/Volumes/ledes/c7source/feeds/packages/libs/p11-kit'
make[2]: *** [package/feeds/packages/p11-kit/compile] Error 2
make[2]: Leaving directory `/Volumes/ledes/c7source'
make[1]: *** [/Volumes/ledes/c7source/staging_dir/target-mips_24kc_musl-1.1.16/stamp/.package_compile] Error 2
make[1]: Leaving directory `/Volumes/ledes/c7source'
make: *** [world] Error 2
```

nmav, 2018-01-29:
It most likely need to be provided the path of ca-certificates bundle or so, but then I wonder why it was working until now. I'll try to investigate next weeked, but if you could check what is going on there would be great help.

hewonen, 2018-01-29:
Does it want the bundle included in the LEDE sources or the host's certificates? Only certificates I could find in macports were for Curl, and just installing them didn't help. I'll see if I run into a solution.

nmav, 2018-01-30:
It seems that this option was needed for the p11-kit trust module, which is not needed/used in openwrt. As such I've disabled it.

commit 4c7a0c31 (2018-01-30): p11-kit: disable trust module

This allows prevents build error due to trust-paths not being
specified. The trust module was not being used in openwrt.

Resolves #5528
commit acc974f8 (2018-01-30): p11-kit: disable trust module

This allows prevents build error due to trust-paths not being
specified. The trust module was not being used in openwrt.

Resolves #5528
closed 2018-01-30 by nmav