openwrt/packages #5538: adblock: Support for blocking on certain local IPs

james-mcguire, 2018-01-30:
All my desktop and laptop already have ad blockers on the browser level.  They often encounter sites that won't load properly and I have to manually disable them.

I want to use the adblock plugin to specifically block ads on my roku / chromecast devices, but no other devices.

In other words, adblock would need to respond to only dns requests from certain devices/IPs.

dibdot, 2018-01-30:
As already explained in the forum it's up to you to configure your dns server settings, e.g. which dns server will be announced via dhcp and which one is only available via static ip configuration - adblock doesn't change any network settings.

Please close this ticket.

james-mcguire, 2018-01-30:
Correct, this is a feature request for that ability... :)  Reviewing the forum, it sounds like several others were interested in this as well.

dibdot, 2018-01-30:
However, it makes no sense. Adblock is only local dns spoofing, no further "automagical" network/firewall stuff. Options for you:

- Use another external dns server without adblock, e.g. 8.8.8.8 (hardcoded or as dhcp default)
- Use multiple local dnsmasq instances, only one with enabled adblock
- Check dns backends like unbound or bind and start working with "tags" or "views"

You will see, the first option is the easiest one.

dibdot, 2018-01-31:
Bottomline this 'feature request' cannot be satisfied. Possible workarounds are illustrated, so I am going to close this one.

closed 2018-01-31 by dibdot