openwrt/packages #5545: Is any work in progress on integrating IoT into generic OpenWrt packages?

ponyatov, 2018-01-31:
I need of COAP, LWM2M and MQTT support in firmware build with OpenWrt for HLR-RM04.

In what stage is current work on IoT protocols support in OpenWrt?
What feed should I use to install wakaama or any full-sized LWM2M device client stub?

karlp, 2018-01-31:
Did you try looking at all? coap and mqtt are both well supported, and have been for years.  I've not looked explicitly for any lwm2m packages, they tend to all be java based, which is normally well out of scope for all the common openwrt target platforms.

- https://github.com/openwrt/packages/tree/master/libs/libcoap
- https://github.com/openwrt/packages/tree/master/net/mosquitto
- https://github.com/openwrt/packages/tree/master/net/iotivity
- just for starters.

closed 2018-01-31 by karlp

ponyatov, 2018-02-01:
can't find wakaama or AwaLWM2M, modern commercial cloud platforms I have to support require lwm2m only. Ok, it's time to contribute...