openwrt/packages #5552: stubby: default configuration is questionable

EnergyMouse, 2018-02-01:
Maintainer: @iamperson347
Environment: LEDE/OpenWrt current

Description:
I find [default config](https://github.com/openwrt/packages/blob/bbab28385b9123465ec46a7cf36178b4aed4dc2b/net/stubby/files/stubby.yml) very questionable: 'round_robin_upstreams' option disabled (enabled in upstream), only Quad9 as upstream server, tls_pubkey_pinset data removed.
Quad9 differs from the usual dns server and should not be offered by default:
- Filtering traffic.
- Has no track record.
- Support DNS-over-TLS unofficially.
- Slow (at least compared to Google and Cisco DNS servers)
- Does not support EDNS Client-Subnet and can resolve far located servers.

iamperson347, 2018-02-01:
Hi,

I changed the default to quad 9 since their servers may be more
accessible/faster for a greater number of people (vs. the defaults). Also,
with regards to the 'round_robin_upstreams' setting, this was to force ipv6
first, and then fall back to ipv4 (since I only listed quad9, I think this
config could make sense).

If people would prefer to revert back to the default configs (upstream), I
have no problem doing so. This was my first stab at getting the package
available for the openwrt community. (Also my first go at an openwrt
package.)

Thank you for the feedback!

iamperson347, 2018-02-02:
Hi @EnergyMouse 

I thought about this some more and do agree with some of your points. What are your thoughts about switching to quad9's non-filtering service?

9.9.9.10
2620:fe::10

This will address the traffic filtering and EDNS Client-subnet concerns (both of which I agree our important).

closed 2018-03-16 by EnergyMouse