openwrt/packages #5554: miniupnpd doesn't keep multiple "listening_ip" setting

sfjuocekr, 2018-02-01:
I added an extra interface to listen on in /var/etc/miniupnpd.conf, sadly it doesn't survive a reboot.

The LUCI app doesn't allow multiple listening interfaces either.

exdeus7, 2018-02-02:
That file is the runtime file generated by the init script after parsing the UCI config.  The Luci app, uci config file and init script need to have support added for this

neheb, 2020-02-09:
Going to close this as it's a fairly rare situation.

To solve: config_file can be specified in /etc/config/upnpd for a custom configuration file. Otherwise, one gets generated for you, The generated file does not handle multiple IPs at the moment.

The listening_ip setting actually is in a loop: https://github.com/openwrt/packages/blob/master/net/miniupnpd/files/miniupnpd.init#L124

Maybe that can give you hints...

closed 2020-02-09 by neheb