openwrt/packages #5555: sqlite3-cli: package should depend on libsqlite3

gbuchetet, 2018-02-01:
The package sqlite3-cli is supposed to depend on libsqlite3:

```
# opkg info sqlite3-cli
Package: sqlite3-cli
Version: 3190300-2
Depends: libc, libsqlite3, libncurses, libreadline
Status: install user installed
Section: utils
Architecture: mips_24kc
Size: 376935
Filename: sqlite3-cli_3190300-2_mips_24kc.ipk
Description: SQLite is a small C library that implements a self-contained, embeddable,
 zero-configuration SQL database engine.
 This package contains a terminal-based front-end to the SQLite (v3.x) library
 that can evaluate queries interactively and display the results in multiple
 formats.
Installed-Time: 1517518932
```
The sqlite3 binary is statically built, though:

```
# ldd /usr/bin/sqlite3 
	/lib/ld-musl-mips-sf.so.1 (0x774c8000)
	libreadline.so.7 => /usr/lib/libreadline.so.7 (0x7748a000)
	libncursesw.so.6 => /usr/lib/libncursesw.so.6 (0x77444000)
	libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x77420000)
	libc.so => /lib/ld-musl-mips-sf.so.1 (0x774c8000)
```
That should save most of space used by the binary.

Regards,
Gaëtan

karlp, 2018-02-01:
Could you try just maybe deleting https://github.com/openwrt/packages/blob/master/libs/sqlite3/Makefile#L94 and see if it fixes it al?

gbuchetet, 2018-02-02:
Removing --enable-static is not sufficient (it is enabled by default), and --disable-static just removes the libsqlite3.a file (which is released in devel package). The magic option is --disable-static-shell (works fine).

See attached patch.

[sqlite3.zip](https://github.com/openwrt/packages/files/1690584/sqlite3.zip)

karlp, 2018-02-02:
If you can make this a proper pull request on thiis repo, it willl be much easier for it get merged

champtar, 2018-02-02:
@gbuchetet was going to say the same as @karlp 
now that you've done most of the work, please finish it by doing a proper PR

gbuchetet, 2018-02-03:
Done.
This is my first PR ever, so let me know if there is something wrong...

mhei, 2018-02-12:
It would be nice, if you:
- could squash both commits into a single one
- move this commit in a branch of your own repository and open a proper PR
- sign-off this commit
- change the first line of the commit message to something like "sqlite3: ... (fixes #5555)"
   and the explain this change in the following lines
- increase PKG_VERSION

(Do not give up, the first time is always the hardest. 👍 )

gbuchetet, 2018-02-12:
What is wrong with PR #5566? I thought that:
- it already refers to a single commit
- the commit is done on a branch from my own repository, based on master, as suggested by @karlp (and not the 16.01 branch)
- the commit is signed-off

I bumped PKG_RELEASE only, because I have not touched the sqlite3 version at all.

Let me know if I need to close #5566 and open a new - and clean - one!

mhei, 2018-02-13:
Ah, sorry, I just did not notice #5566 but only looked at the commits referenced above (and additionally, I confused PKG_RELEASE with PKG_VERSION). Your PR looks fine and it seems that @karlp will take care if it. Sorry about my noise here...

gbuchetet, 2018-02-13:
No problem!

karlp, 2018-02-19:
Merged in 572a717274e12505b2bd6659ef48fe33d3a3ca6e sorry it took me so long to get back to this!

closed 2018-02-19 by karlp