openwrt/packages #5559: autossh work if manuel /etc/init.d/autossh start, but not on reboot, ssh exit error status 1

winfriedmuench, 2018-02-02:
tested with WRT54GS V2 and netgear WA801DN
LEDE 17.01.4 r3560-79f57e422d

If I login with ssh end run
```
/etc/init.d/autossh stop
/etc/init.d/autossh start
```
it works well.

After reboot, loop with
in syslog
```
user.info autossh[pid]: ssh exited with error status 1; restarting ssh 
```
after 5 seconds the same
No Idea

winfriedmuench, 2018-02-05:
Also restart over system/startup works.

winfriedmuench, 2018-02-05:
I tryed to move the private key to /etc/dropbear/id_dropbear. Its the same.
Tryed also:
```root@LEDE:/etc# cat rc.local 
```# Put your custom commands here that should be executed once
```# the system init finished. By default this file does nothing.
```sleep 180
```/etc/init.d/autossh restart
```exit 0
```root@LEDE:/etc#

same thing. AAAARG

dibdot, 2018-02-05:
@bk138

bk138, 2018-02-05:
I can just say it works for me... Would you mind posting your config @winfriedmuench ?

winfriedmuench, 2018-02-05:
sure
the hole one:
[backup-LEDE-2018-02-05.tar.gz](https://github.com/openwrt/packages/files/1695620/backup-LEDE-2018-02-05.tar.gz)

bk138, 2018-02-05:
Thank you! From a first glance, the `/443 ` in /etc/config/autossh looks weird to me. Have you tried running ssh with those args from the cmdline, with autossh stopped?

winfriedmuench, 2018-02-05:
Yes. It works perfect if i login with ssh and do /etc/init.d/autossh restart
i need 443 because of some firewall rules in other environments.

bk138, 2018-02-05:
I am not aware that ssh understand this slash syntax. Running `ssh -i /etc/dropbear/id_dropbear -N -T -R 5558:192.168.1.1:22 rlogin@asteriskpi1.dyn-ip24.de/443` (from you /etc/config/autossh) gives an error about the hostname.

winfriedmuench, 2018-02-05:
but it works well if startet on shell. The slash is implemented in the dropbear ssh. Have try it also with -p option. You are welcome to try it on my server. Key should be includet. Login not possible but remote port forwarding possible.

bk138, 2018-02-05:
Mhh, that might be a hint. I am using openssh-client. What do `which ssh` and `ssh -V` give you?

winfriedmuench, 2018-02-05:
[image]

winfriedmuench, 2018-02-06:
next hint I hope, my internet connection is over WLAN. 
And my Server auth.log:
```
Feb  6 09:05:54 asteriskpi1 sshd[20573]: fatal: Read from socket failed: Connection reset by peer [preauth]
```
But when I start autossh manual for example on Web/System/Startup click on autossh restart:
```
Feb  6 11:25:21 asteriskpi1 sshd[21078]: Accepted publickey for rlogin from 79.221.4.19 port 43006 ssh2: RSA e6:fe:5f:b6:c0:f3:46:6e:19:e8:aa:fa:15:9a:ba:54
Feb  6 11:25:21 asteriskpi1 sshd[21078]: pam_unix(sshd:session): session opened for user rlogin by (uid=0)
Feb  6 11:25:21 asteriskpi1 systemd-logind[296]: New session c626 of user rlogin.
```

winfriedmuench, 2018-02-06:
...and there is no differenc between keyfile in /root/.ssh/ or /etc/dropbear/

winfriedmuench, 2018-02-06:
...and yes. I verified, after install openssh-client package and modify the config from /443 to -p 443 and use a openssh keyfile, it works after reboot fine. After remove openssh-client, and start with "/etc/init.d/autossh start:
```
Tue Feb  6 12:06:25 2018 user.info autossh[1158]: ssh exited with error status 1; restarting ssh
```
because the keyfile is in openssh format. Change back to the key from dropbear and after stop and start it works. After reboot the same error. Seems to me that the id_dropbear ist not accessible while boot? Or there is some service that enabled the access to the dropbear keyfile?

by the way. using of openssh-client is no solution,because is to big for some 4MB Flash router of mine.

winfriedmuench, 2018-02-06:
In the past I tryed in rc.local:
```
sleep 60 
/etc/init.d/autossh restart
```
nothing, but with
```
. /etc/profile
/etc/init.d/autossh restart
```
it works. output from set befor . profile and after: 
```
root@LEDE:/tmp# diff environment.rc.local*
1c1
< HOME='/'
---
> HOME='/root'
8c8
< PS1='\w \$ '
---
> PS1='\u@\h:\w\$ '
```
So PS1 should not be anything, but HOME?, try now in rc.local
```
export HOME=/root
```
and YES it works. But why? Perhaps known_hosts file in $HOME/.ssh/ ??? 
I try to add in /etc/config/autossh in ssh the parameter -o 'UserKnownHostsFile=/root/.ssh/known_hosts' dosn't work. Ah, is not compiled:
```
root@LEDE:~# ssh -o help
ssh: Available options:
	ExitOnForwardFailure
	UseSyslog

root@LEDE:~# 
```
other solution I testet to add in /etc/init.d/autossh:
```
export HOME=/root
```
this works and is my propose.

winfriedmuench, 2018-02-07:
Or is this a bug that $HOME is / douring boot?

bk138, 2018-02-08:
@winfriedmuench I proppose you try to isolate the issue and try starting a single ssh session without autossh the way you do it. Thus you can see if autossh is the culprit or if it's another issue.

winfriedmuench, 2018-02-08:
It was the first thing befor I open the issue to try ssh without autossh. Yes I used rc.local to isolate the issue, first to automatic restart autossh after some sleep with /etc/init.d/autossh restart. And I write the environment with set > /tmp/environment.rc.local.txt in a file.

winfriedmuench, 2018-02-08:
The colprit is that dropbear ssh seek the known_hosts file in $HOME/.ssh/known_hosts. There is no Parameter to change this. If $HOME=/ there is no known_hosts and no connection possible. if $HOME=/root evrything is OK.

bk138, 2018-02-08:
I guess you should then file an issue with the dropbear package instead.

winfriedmuench, 2018-02-08:
OK, then I close this issue.

closed 2018-02-08 by winfriedmuench

daztucker, 2018-10-13:
You can also work around it by symlinking /root/.ssh /.ssh