openwrt/packages #5573: boost : compilation fail with target-x86_64_glibc

atyg02, 2018-02-05:
Maintainer: @ClaymorePT 
Environment: Environment: x86_64, OpenWRT/LEDE trunk
Description: Compilation failed with target-x86_64_glibc
[Build.log](https://github.com/openwrt/packages/files/1695623/Build.log)

ClaymorePT, 2018-02-05:
@atyg02  I have executed a compile test using the following configurations
LEDE Project with GCC 5.x
I have selected Boost.Thread to check your issue and no issues have appeared. The package was compiled properly.

Could you test it again in a clean-slate environment?

atyg02, 2018-02-06:
@ClaymorePT
I have tested with GCC 5.5.0.
```
# ------------------------------------
# Target System                                              --> x86
# Subtarget                                                     --> x86_64
# Target Profile                                               --> Generic
```
if I select only --> <*> boost  -- it compiled with no issues.

But if I try to include  the following Boost library, then failed to compile.
```
--> <*> boost.................................. Boost C++ source library packages
                       --> Select Boost libraries
                                            --> -*- Boost atomic library   
                                            --> -*- Boost chrono library
                                            --> -*- Boost system library
                                            --> -*- Boost thread libray
```
It tested with the following target and compiles with no issues.
```
# ------------------------------------
# Target System                     --> Marvell Armada 37x/38x/XP
# Subtarget                            --> (Linksys WRT1200AC (Caiman)
```

 Thank you !

ClaymorePT, 2018-02-06:
Well, I tested with the exact configuration you just provided for x86_64, and it compiled flawlessly. 
If you are changing targets, please execute a `make clean` before compiling. It usually avoids having issues.

atyg02, 2018-02-06:
@ClaymorePT 
tested with a fresh clone from git repository, however still complains.
As I have explicitly wrote in the title I used **glibc toolchain**, the default is **musl** .
with which toolchain have tested ( glibc or musl) ?

Thank you for your patience.

ClaymorePT, 2018-02-06:
I have tested using musl.

I will go test it with glibc then, to replicate the issue.

ClaymorePT, 2018-02-06:
@atyg02  I can now confirm your issue. I will work it as soon as I can.

atyg02, 2018-02-06:
@ClaymorePT,  Thank you ! and I appreciate your support.

ClaymorePT, 2018-02-09:
@atyg02  It seems that the issue you detected, is [a regression introduced by glibc 2.26](https://git.busybox.net/buildroot/commit/?id=d6a4807f9473b4322e96b77169ffcd6d9cd84170)

There's nothing I can do from this side. It would be better to submit an issue report for glibc, and reference this issue here.

ClaymorePT, 2018-02-09:
@atyg02 After doing some digging and experimentation, it seems that [the blame falls over glibc 2.26](https://git.busybox.net/buildroot/commit/?id=d6a4807f9473b4322e96b77169ffcd6d9cd84170).
glibc 2.26 has introduced a regression and needs to be patched. 
This issue is not related to boost and because of that, from this side I cannot do anything.

The reason why building the packege works when you just select _boost<*>_ is because by doing so, libraries that required compilation will not be included. Using the just _boost<*>_ will copy all of the header-libs-only to the cross-compiler staging dir, so that you can develop applications, using only the header-libs from boost.
So, _boost<*>_ does not trigger the compilation of Boost.Thread, which is the package that causes that error to be triggered. 

[According to the patch submission from buildroot](https://git.busybox.net/buildroot/commit/?id=d6a4807f9473b4322e96b77169ffcd6d9cd84170) the issue appears for **x86, x86_64 and
powerpc64le** only. That is why it works for other targets.

You should open an issue related to glibc 2.26 and reference this issue there.
Regarding this issue, like I said, there's nothing I can do. 
If you have any other issues, let me know.

atyg02, 2018-02-09:
@ClaymorePT , Thank you for your time and effort.

diizzyy, 2018-07-05:
Does it still fail?

ClaymorePT, 2018-07-05:
@diizzyy Just tested a build. Everything went out just fine.

closed 2018-07-06 by yousong