openwrt/packages #5583: dropbear ssh no parameter for location of known_hosts file

winfriedmuench, 2018-02-08:
so I try to use autossh. It works well, but not on reboot. Diggin around found on reboot $HOME is / 
dropbear search known_hosts in $HOME/.ssh/known_hosts. There is no parameter to change that:
```
root@LEDE:~# ssh -o help
ssh: Available options:
	ExitOnForwardFailure
	UseSyslog

root@LEDE:~# ssh -V
Dropbear v2017.75
root@LEDE:~# #
```
would be nice to integrate parameter UserKnownHostsFile or GlobalKnownHostsFile. Or make shure that on boot $HOME is set to /root.
See original issue on [autossh](https://github.com/openwrt/packages/issues/5559#issuecomment-364055154)

dibdot, 2018-02-13:
seems to be a race condition in autossh package during boot. In my (booted) environment HOME always points to '/root'. I think it doesn't make sense to hardcode dropbear (which is not maintained at the package repo!) to use this directory. A fix should be implemented in autossh, e.g. switch init to procd, use network trigger and explicitly set the home dir environment variable or change the existing hotplug script.

@winfriedmuench please always add the current package maintainer to your issues 

for reference: @bk138 

Anyway I close this by now cause dropbear is not part of this repo. Issues with core components are tracked here: https://bugs.openwrt.org/

closed 2018-02-13 by dibdot

B0F1B0, 2024-05-21:
My content of `ls -all /root/.ssh/`
````
-rw-------    1 root     root            83 May 20 10:11 id_dropbear
-rw-r--r--    1 root     root           186 May 20 10:09 known_hosts
````
For me the following works 😎

````
cp /root/.ssh/* /.ssh/
````

````
service dropbear restart
````