openwrt/packages #5586: logrotate: not scheduled from crond

pprindeville, 2018-02-09:
Maintainer: @bk138 
Environment: x86_64, generic, HEAD

Description:

I've installed `logrotate` but it never gets run.  On inspection, it seems to be because it's not being dispatched nightly from `/etc/cron.daily/` and indeed, `opkg files logrotate` verifies that it's not installing a hook for `crond`.

Looking at `crond` also, I'm seeing that it's:

- failing to start at boot-time
- not logging to syslog

Further, it (`crond`) fails to indicate any sort of error, it just silently fails to start.

I'll open a separate bug for the latter.

Is there a reason that the file `logrotate-n.n.n/examples/logrotate.cron` can't be used?

bk138, 2018-02-10:
Yes, the current logroate is provided bare-bones, without any default config. This is intentional, as sensible defaults are hard to find (for instance, I'd say it ought to be scheduled hourly, but YMMV...).

pprindeville, 2018-02-12:
I can live with hourly.

bk138, 2018-02-12:
PRs always welcome!

jow-, 2018-02-12:
crond only starts at boot time of crontabs are installed.

feckert, 2019-05-25:
We have now micrond https://github.com/openwrt/packages/blob/master/utils/micrond/Makefile

ilario, 2019-10-21:
Maybe adding something like this to the Makefile could fix?

```
define Package/logrotate/postinst
#!/bin/sh
if [ -z "$${IPKG_INSTROOT}" ]; then
	(crontab -l ; echo "0 8 * * * *     /usr/sbin/logrotate /etc/logrotate.conf") | sort | uniq | crontab -
fi
exit 0
endef

define Package/logrotate/prerm
#!/bin/sh
if [ -z "$${IPKG_INSTROOT}" ]; then
	crontab -l | grep -v "0 8 * * * *     /usr/sbin/logrotate /etc/logrotate.conf" |  sort | uniq | crontab -
fi
exit 0
endef
```

The proposed solution was copied partially from `fdm` and from `wifischedule` packages.

One possible remaining problem is that currently logrotate.status is in RAM (`/var/lib/logrotate.status`) and gets lost when rebooting.

feckert, 2019-10-25:
This will dirty the view in the crontab.
At some point you installed things and then you don't know who wrote what.
We do not have the full crontab only the busybox crontab and this should not be changed in the future?
So I added some time ago the package https://github.com/openwrt/packages/tree/master/utils/micrond from freifunk thanks to @NeoRaider.
Can we not solve system relevant time tasks in this way?

neheb, 2020-02-09:
Maintainer says this is intentional. Closing.

closed 2020-02-09 by neheb