openwrt/packages #5591: prometheus-node-exporter-lua: memory leak

tyriis, 2018-02-10:
Maintainer: @champtar 
Environment: (yeah i know its not openwrt, but i dont think this is the reaseon for a mamory leak) LEDE Reboot 17.01.4 r3560-79f57e422d / LuCI lede-17.01 branch (git-17.290.79498-d3f0685)

Description: i observe a memory leak in prometheus-node-exporter-lua, memory consummation increase until process is killed/restarted/dies. (+3.75 MB/h). I have tried different versions, the released package and also the latest code version, they all have the same issue.

champtar, 2018-02-11:
Hi @tyriis 
I'll try to have a look, but if you have some time, can you try to scrape only some collector? Without looking at the code, I'm suspecting WiFi collector (ubus misuse?)

tyriis, 2018-02-12:
@champtar my bad, it seems that i have had an old Version running. Everything looks good now.

closed 2018-02-12 by tyriis