openwrt/packages #5615: netifd: "ubus call network reload" can't remove the deprecated ipv6 prefix delegation on lan

alfred1982, 2018-02-14:
# Issue template 

Maintainer: Hans Dedecker <dedeckeh@gmail.com>
Environment: OpenWrt Chaos Calmer 15.05.1, netifd-0f96606b7040b8e14190ff055d5761744bc15f6d

Description:
"ubus call network reload" can't remove the deprecated ipv6 prefix delegation on lan interface

1. I create a wan connection of type DHCPv6 with reqiapd enabled. Call "ubus call network reload"
2. When wan connection is up, my lan interface br-lan will get delegated ipv6 address
3. I remove the DHCPv6 wan connection and create a static IPv4 interface.  Call "ubus call network reload"
4. The delegated ipv6 address is still there after step 3.

"ubus call network restart" can fix this, but it is not preferred, for it will restart other unchanged interfaces too.
This seems to be a bug in netifd. My netifd release is 0f96606b7040b8e14190ff055d5761744bc15f6d. 

Thanks.

dibdot, 2018-02-18:
Please check if the mentioned issue is still present in current releases 17.01.x or current snapshot. Anyway netifd is not part of the package repo. If this issue still exist for you, please open a ticket here: https://bugs.openwrt.org/

closed 2018-02-18 by dibdot

alfred1982, 2018-02-27:
Thanks for the reply. 
I've opened a ticket at bugs.openwrt.org.