openwrt/packages #5616: netdata: Install on OpenWRT

sereeds, 2018-02-14:
It appears that "netdata" is no longer found in the official OpenWRT packages for easy install.
Are the ipk files still being maintained for netdata?

https://github.com/openwrt/packages/tree/master/admin/netdata

karlp, 2018-02-14:
where are you looking? http://downloads.openwrt.org/snapshots/packages/mips_24kc/packages/netdata_1.9.0-1_mips_24kc.ipk for instance has it just fine, built this morning.

sereeds, 2018-02-14:
I was using the opkg command line tool on OpenWRT.
I see it looks for packages here:
http://downloads.openwrt.org/chaos_calmer/15.05.1/ar71xx/generic/packages/packages/

For example I can add the "htop" package, but not netdata.
opkg update
opkg list htop
  (gives result)
opkg search *htop*
  (gives result)
opkg search *netdata*
  (no results)

Is there somewhere else I should look for a netdata ipk for my ar71xx device?
I assumed it was provided by the default package repos just like htop is.

Thanks.

sereeds, 2018-02-14:
So I updated my device from 15.05, to the new LEDE 17.01.4.
Now thew netdata package installs as it should, from the opkg.

So upgrading to latest stable version of OpenWRT/LEDE fixed my issue.
Thanks.

closed 2018-02-14 by sereeds

sereeds, 2018-02-14:
I just noticed that it installed netdata version 1.4, and 1.9 is the latest.
Is there any way to get that package updated in the repo?
http://downloads.lede-project.org/releases/17.01.4/packages/mips_24kc/packages/

I was able to manually update to version netdata 1.9 using the link that you provided above.
http://downloads.openwrt.org/chaos_calmer/15.05.1/ar71xx/generic/packages/packages/
Installing from that location works now that I'm at LEDE 17.01.

So I was able to get the netdata version 1.9 working on LEDE 17.01.
Thanks.