openwrt/packages #5628: transmission: setting ulimit values for transmission-complete script to open applications

SimpleBobster, 2018-02-17:
Maintainer: @\neheb 
Environment: WRT1900ACS, LEDE Reboot 17.01.4 r3560-79f57e422d / LuCI lede-17.01 branch (git-17.290.79498-d3f0685)

Description:
I have rclone running on a script that is called after a transmission download. In the log for the script, I keep getting virtual memory errors when I try to call it ([documented here](https://forum.rclone.org/t/potential-bug-when-running-rclone-off-openwrt/4726)

Of course, I run the transmission-daemon with the default transmission user, so setting ulimit -v in the script to unlimited gets an "operation not permitted error" ([documented here](https://forum.transmissionbt.com/viewtopic.php?f=2&t=18916&p=76032&sid=22936f75a089fa1de34e3cb3f6fa100d#p76032)

I can't set the ulimit in /etc/security since that doesn't exist, ([documented here] (https://forum.lede-project.org/t/setting-ulimit-values-without-etc-security-limits-conf/11731)] so my only option seems to be changing the /etc/init.d/transmission startup script, which is maintained here. I don't feel comfortable changing that script. 

Is there any way to set the ulimit -v value to unlimited in transmission's init.d so that I can run rclone from the script torrent done? What is the harm in setting it that way by default?

neheb, 2018-02-17:
Set mem_percentage in /etc/config/transmission to 100 and try again.

I have no idea why this setting exists. A different user reported on a 32MB device that he needed to do this or transmission would segfault.

I am considering removing it.

SimpleBobster, 2018-02-17:
Thanks. That fixed it––rclone runs successfully. I did a little digging and it seems it came from this [commit]
(https://github.com/openwrt/packages/commit/6091831a5b6257f79a42bc5ed6591f89c127752b#diff-f699bc44f3528b9747203af2c274dc13)

Seems to come from concern over transmission stability. Not sure why they had to include a memory limiter though? Doesn't unix manage that internally?

https://github.com/openwrt/packages/pull/1363

neheb, 2018-02-17:
It does but the vm manager needs to be tweaked. I'll look into it when I have time.

neheb, 2019-03-30:
I removed the memory limit a while back. That eliminates the main issue. Closing.

closed 2019-03-30 by neheb