openwrt/packages #5630: usbip: breaks with kernel 4.14

Borromini, 2018-02-18:
Maintainer: @nunojpg 
Environment: OpenWrt master, x86/64 with kernel 4.14 manually selected.

Description: Compilation breaks with kernel 4.14

```
make[6]: Entering directory '/var/data/code/lede/master/build_dir/target-i386_pentium4_musl/linux-x86_generic/usbip-2.0/src'
  CC       usbip_list.o
usbip_list.c: In function 'list_devices':
usbip_list.c:214:11: error: assignment makes pointer from integer without a cast [-Werror=int-conversion]
   devpath = udev_device_get_devpath(dev);
           ^
cc1: all warnings being treated as errors
```

naps1saps, 2018-05-31:
I'm not following NeoRaider.  Was there a temporary fix for `error: assignment makes pointer from integer without a cast` or is that still pending?

NeoRaider, 2018-07-02:
The current version of usbip userspace does not work with out minimal libudev shim anymore. The eudev project is dead, so I see two options:
1. Patch usbip to work with our libudev replacement again (might re-introduce bugs in usbip whose fixes cause the incompatiblity... and the patches might break often, as the usbip sources are bundled with the kernel)
2. Someone would need to package full udev (which is part of systemd). systemd has switched to the Meson build system a while ago, which we don't support at the moment (and which requires ninja, which is not packaged either). A lot of Yak shaving... but the effort might pay off, as I'm also interested in Meson support for other packages. Unfortunately, I'm somewhat busy at the moment, so unless someone steps up to maintain all those tools, it will not happen any time soon...

arrmo, 2018-08-02:
Hi,

This seems to be fixed with this recent commit,
https://github.com/openwrt/openwrt/commit/4f3082583a0f08bf8329c2a6e754fbf4e0bea544

But ... having issues like this now - thoughts?
```
root@wap-garage:~# usbip list -l
udev_new
udev_enumerate_new
udev_enumerate_add_match_subsystem
stub: udev_enumerate_add_nomatch_sysattr bDeviceClass 09
stub: udev_enumerate_add_nomatch_sysattr bInterfaceNumber (null)
udev_enumerate_scan_devices
udev_enumerate_unref
udev_unref
```

```
root@wap-garage:~# usbip bind -b 1-1.4
udev_new
udev_device_get_syspath
udev_unref
udev_new
stub: udev_device_get_sysattr_value bDeviceClass
udev_device_unref 0x77f23440 0
udev_unref
usbip: error: could not unbind driver from device on busid 1-1.4
```
Thanks!

Dimazhan, 2018-09-01:
Hi!

I tested usbip with eudev, it works.
my patch:
[libudev-eudev.patch.gz](https://github.com/openwrt/packages/files/2342075/libudev-eudev.patch.gz)
I tested for ar71xx (TL-MR3020), openwrt-18.06, kernel 4.9

c-MM, 2018-09-03:
Thanks a lot for your patch. I got usbip working with it on 18.06.1 ar71xx-generic for an Tplink wdr4300 using the SDK to recompile the packages. It would be nice if openwrt would find a way to make it work again on the original images.

lexxai, 2018-09-07:
c-MM : Can some where post how you do it for 18.06.1 ar71xx-generic for an Tplink wdr4300?
Need replace libudev-fbsd by libudev-eudev (https://github.com/gentoo/eudev.git)?

neheb, 2020-01-12:
Is this still valid?

nunojpg, 2020-02-02:
@Borromini please comment if this is still valid,

closed 2020-02-02 by nunojpg