openwrt/packages #5631: jool: compilation breaks with 4.14

Borromini, 2018-02-18:
Maintainer: @danrl 
Environment: OpenWrt master, x86/64 with kernel 4.14

Description:
Compiling jool breaks on OpenWrt master with kernel 4.14.

```
/var/data/code/lede/master/build_dir/target-i386_pentium4_musl/linux-x86_generic/jool-997a81bb5f5e9d82aa122fd37b7c890e44a245dd/mod/stateful/nf_hook.c: In function 'jool_init':
/var/data/code/lede/master/build_dir/target-i386_pentium4_musl/linux-x86_generic/jool-997a81bb5f5e9d82aa122fd37b7c890e44a245dd/mod/stateful/nf_hook.c:155:10: error: implicit
declaration of function 'nf_register_hooks' [-Werror=implicit-function-declaration]
  error = nf_register_hooks(nfho, ARRAY_SIZE(nfho));
          ^
/var/data/code/lede/master/build_dir/target-i386_pentium4_musl/linux-x86_generic/jool-997a81bb5f5e9d82aa122fd37b7c890e44a245dd/mod/stateful/nf_hook.c: In function 'jool_exit':
/var/data/code/lede/master/build_dir/target-i386_pentium4_musl/linux-x86_generic/jool-997a81bb5f5e9d82aa122fd37b7c890e44a245dd/mod/stateful/nf_hook.c:187:2: error: implicit declaration of function 'nf_unregister_hooks' [-Werror=implicit-function-declaration]
  nf_unregister_hooks(nfho, ARRAY_SIZE(nfho));
  ^
```

danrl, 2018-02-21:
@NICMx @ydahhrk Do we have new dependencies when using Kernel 4.x?

ydahhrk, 2018-02-22:
I see that you already fixed the problem, but just to clarify:

@danrl No, but support for kernels 4.13 and 4.14 was added only recently in Jool 3.5.6.

danrl, 2018-02-23:
Seems to be fixed in #5654