openwrt/packages #5633: mosquitto: libmosquitto-ssl should depend on CONFIG_OPENSSL_WITH_PSK=y

nwf, 2018-02-19:
Maintainer: @karlp 
Environment: LEDE reboot-6185-g16245a5d8e
Description: 

Failing to capture this dependency gives rise to this build failure when attempting to build without PSK functions in OpenSSL:
```
make[5]: Entering directory '/tank/openwrt/scratch/builder-usbnetgw2/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/mosquitto-ssl/mosquitto-1.4.14/client'
ccache_cc pub_client.o client_shared.o -o mosquitto_pub -L/tank/openwrt/scratch/builder-usbnetgw2/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/lib -L/tank/openwrt/scratch/builder-usbnetgw2/staging
_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/lib -L/tank/openwrt/scratch/builder-usbnetgw2/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-5.5.0_musl_eabi/usr/lib -L/tank/openwrt/scratch/builder-usbnetgw
2/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-5.5.0_musl_eabi/lib -specs=/tank/openwrt/scratch/builder-usbnetgw2/include/hardened-ld-pie.specs -znow -zrelro  -L../lib ../lib/libmosquitto.so.1
../lib/libmosquitto.so.1: undefined reference to `SSL_CTX_set_psk_client_callback'                              
collect2: error: ld returned 1 exit status                                                                     
Makefile:8: recipe for target 'mosquitto_pub' failed                                                                  
make[5]: *** [mosquitto_pub] Error 1
```

karlp, 2018-02-19:
Yes, this is what happens when people add mountains of config options.  If you want me to fix this, it's going to be a hard dep on PSK, I'm not going to try and keep playing this game of ever multiplying configs and allow you to configure mosquitto without psk support.

karlp, 2018-02-19:
turns out it was just as easy to make this magical, and auto include tls-psk based on the openssl setting.  merged in 055be639b385029a7540ec22cc926392c7ca1dde thanks

closed 2018-02-19 by karlp