openwrt/packages #5637: libiio: missing dependencies

stintel, 2018-02-19:
Maintainer: @mhei 
Environment: OpenWrt master r6192-8651f1149d

Description:

```
Package libiio is missing dependencies for the following libraries:
libavahi-client.so.3
libavahi-common.so.3
```

mhei, 2018-02-20:
@stintel could you provide more details, e.g. which platform and which git commit of packages etc.?
I thought this issue was already fixed and can not reproduce it here.

stintel, 2018-02-22:
brcm2708/bcm2708, packages @ 055be639b385029a7540ec22cc926392c7ca1dde

mhei, 2018-02-26:
I can reproduce this issue only, when I first select libavahi-client package, then manually deselect it and rebuild libiio. Then the avahi library and headers are still present in the staging directories and libiio find/use it which triggers the missing deps errors...
I wonder whether I should make the deps unconditionally (which always increases the footprint), or add a config item for this...

stintel, 2018-02-27:
I've looked at the code and iirc there is no cmake option to disable the libavahi dependency. I think it would be best to add such an option, and disable it when libavahi-* is not enabled. I might do it myself, but didn't find the time yet.

commit 457594fd (2018-03-04): libiio: rework backend selection (fixes #5637)

Provide configuration items to select individual backends. This
also allows to fine-tune the package dependencies.
In order to address #5637, we make this dep unconditionally
until upstream allow to opt-in/opt-out libavahi support.
closed 2018-03-04 by mhei