openwrt/packages #5638: Python-2.7.14: host build of python segfaults on macOS 10.13.3

jpc, 2018-02-20:
Maintainer: @commodo
Environment: LEDE, branch master
Package: python2
Description:

After fixing #5310 python builds but segfaults when running `python -E -m ensurepip`. Running under `lldb` suggests a problem with _ctypes:
```
[jpc@jpc OpenWRT] lldb -- ./staging_dir/hostpkg/bin/python -E -m ensurepip
(lldb) target create "./staging_dir/hostpkg/bin/python"
Current executable set to './staging_dir/hostpkg/bin/python' (x86_64).
(lldb) settings set -- target.run-args  "-E" "-m" "ensurepip"
(lldb) r
Process 9373 launched: './staging_dir/hostpkg/bin/python' (x86_64)
Process 9373 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0x2fffc0)
    frame #0: 0x00007fff73a647ca libffi.dylib`ffi_prep_closure + 14
libffi.dylib`ffi_prep_closure:
->  0x7fff73a647ca <+14>: movw   $0xbb49, (%rdi)           ; imm = 0xBB49
    0x7fff73a647cf <+19>: leaq   -0x736(%rip), %rax        ; ffi_closure_unix64
    0x7fff73a647d6 <+26>: movq   %rax, 0x2(%rdi)
    0x7fff73a647da <+30>: movw   $0xba49, 0xa(%rdi)        ; imm = 0xBA49
Target 0: (python) stopped.
(lldb) bt
* thread #1, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0x2fffc0)
  * frame #0: 0x00007fff73a647ca libffi.dylib`ffi_prep_closure + 14
    frame #1: 0x0000000105350ce5 _ctypes.so`_ctypes_alloc_callback + 549
    frame #2: 0x000000010534bbbc _ctypes.so`PyCFuncPtr_new + 268
[...]
```

This may be related to these warnings during compilation:
```
gcc -fno-strict-aliasing -O2 -I/Volumes/OpenWRT/staging_dir/host/include -I/Volumes/OpenWRT/staging_dir/hostpkg/include -I/Volumes/OpenWRT/staging_dir/target-mips_24kc_musl/host/include -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -I_ctypes/darwin -I/usr/include/ffi -I/Volumes/OpenWRT/build_dir/hostpkg/Python-2.7.14/Mac/Include -I. -IInclude -I./Include -I/Volumes/OpenWRT/staging_dir/host/include -I/Volumes/OpenWRT/staging_dir/hostpkg/include -I/Volumes/OpenWRT/build_dir/hostpkg/Python-2.7.14/Include -I/Volumes/OpenWRT/build_dir/hostpkg/Python-2.7.14 -c /Volumes/OpenWRT/build_dir/hostpkg/Python-2.7.14/Modules/_ctypes/callbacks.c -o build/temp.macosx-10.4-x86_64-2.7/Volumes/OpenWRT/build_dir/hostpkg/Python-2.7.14/Modules/_ctypes/callbacks.o -DMACOSX
/Volumes/OpenWRT/build_dir/hostpkg/Python-2.7.14/Modules/_ctypes/callbacks.c:27:9: warning: implicit declaration of function 'ffi_closure_free' is invalid in C99 [-Wimplicit-function-declaration]
        ffi_closure_free(self->pcl_write);
        ^
/Volumes/OpenWRT/build_dir/hostpkg/Python-2.7.14/Modules/_ctypes/callbacks.c:27:9: warning: this function declaration is not a prototype [-Wstrict-prototypes]
/Volumes/OpenWRT/build_dir/hostpkg/Python-2.7.14/Modules/_ctypes/callbacks.c:429:20: warning: implicit declaration of function 'ffi_closure_alloc' is invalid in C99 [-Wimplicit-function-declaration]
    p->pcl_write = ffi_closure_alloc(sizeof(ffi_closure),
                   ^
/Volumes/OpenWRT/build_dir/hostpkg/Python-2.7.14/Modules/_ctypes/callbacks.c:429:20: warning: this function declaration is not a prototype [-Wstrict-prototypes]
/Volumes/OpenWRT/build_dir/hostpkg/Python-2.7.14/Modules/_ctypes/callbacks.c:429:18: warning: incompatible integer to pointer conversion assigning to 'ffi_closure *' (aka 'struct ffi_closure *') from
      'int' [-Wint-conversion]
    p->pcl_write = ffi_closure_alloc(sizeof(ffi_closure),
                 ^ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
5 warnings generated.
```

commodo, 2018-02-20:
@jpc 
well, atm, i don't have a Mac to test / investigate ;

maybe try to switch `--with-system-ffi=to`  to  `--with-system-ffi` in `HOST_CONFIGURE_ARGS` ?
seem `libffi/host` is added to `HOST_BUILD_DEPENDS`, so this could work [if this is the issue]

jpc, 2018-02-22:
It turns out it is a bit more subtle than that. The `setup.py` in the Python distribution has a bug 
and effectively ignores the `--with-system-ffi` flag. More details in #5661

commodo, 2018-02-22:
[curios] what if you remove the `--with-system-ffi` flag altogether ?

jpc, 2018-02-22:
I am pretty sure I have tried that without success. As far as I understand it the configure script normalizes the variable so it is always defined with either a "yes" or a "no" value:
https://github.com/python/cpython/blob/2.7/configure.ac#L2460-L2463
https://www.gnu.org/software/autoconf/manual/autoconf-2.60/html_node/External-Software.html

commodo, 2018-02-23:
I agree that the configure script normalizes yes/no, but in the configure script the variable is `with_system_ffi="no"` whereas in the `setup.py` script it's `'--with-system-ffi`
[ the difference is between dashes and underscores ]

Which is why I am a bit confused.
And in the [generate configure script that's on the cpython repo](https://github.com/python/cpython/blob/2.7/configure#L9341), it looks like `with_system_ffi` is set to no.

But this looks like one of those things that `setup.py` messes up.
Maybe let's do both conditions:
```
        if not '--with-system-ffi' in sysconfig.get_config_var("CONFIG_ARGS"):
               return
        if '--with-system-ffi=no' in sysconfig.get_config_var("CONFIG_ARGS"):
               return
```

That would at least keep some of the old behavior if it's any good, and also cover for the new.
All-in-all this patch may be a good idea to send to `cpython` [if you want].

jpc, 2018-03-12:
Sorry for the long delay but I got swamped with other work. :( I have updated the pull request with the required sign off but feel free to reject it if you are unsure about the patch.

 I can try to get back to it in a little while and maybe try to ask about it on the cpython list.

commodo, 2018-03-14:
Hey,

So, I took a look at the matrix of `--with-system-ffi`.
What I get is:
* CONFIGURE_ARGS = --with-system-ffi=no
   * `'--with-system-ffi=no' in sysconfig.get_config_var("CONFIG_ARGS")` == True
   * `'--with-system-ffi' in sysconfig.get_config_var("CONFIG_ARGS")`       == True
* CONFIGURE_ARGS = --with-system-ffi
   * `'--with-system-ffi=no' in sysconfig.get_config_var("CONFIG_ARGS")` == False
   * `'--with-system-ffi' in sysconfig.get_config_var("CONFIG_ARGS")`       == True
* CONFIGURE_ARGS = <no --with-system-ffi defined>
   * `'--with-system-ffi=no' in sysconfig.get_config_var("CONFIG_ARGS")` == False
   * `'--with-system-ffi' in sysconfig.get_config_var("CONFIG_ARGS")`       == False

So, maybe just removing that from the `HOST_CONFIGURE_ARGS` would be sufficient.
I'll probably just do that.

Thanks for finding this.
After the PR is submitted, maybe can you try to check again on MacOS X ?

Thanks
Alex

commodo, 2018-03-14:
hey,

can you please test with the changes in this PR ?

thanks
Alex

commodo, 2018-03-14:
PR is https://github.com/openwrt/packages/pull/5758

commodo, 2019-02-26:
@jpc 
Python has upgraded since then.
there is issue #7171 that handles some build errors on OS X

At some point, I may have to get a Mac.... :p

commodo, 2019-02-26:
if no objections, can we close this ?
and move to the new issue;

@hnyman 
thoughts ?

closed 2019-02-26 by hnyman