openwrt/packages #5649: Shadowsocks-libev:Shows running anyway.

MisterLin1995, 2018-02-21:
Maintainer: @\<yousong> (find it by checking history of the package Makefile)
Environment: (put here arch, model, OpenWrt version)
[image]

[image]

Description:
The  shadowsocks-libev installed on my openwrt router is not running.
[image]

```
Formating code blocks by wrapping them with pairs of ```
```

dibdot, 2018-02-26:
@yousong

yousong, 2018-02-27:
@MisterLin1995 , have you tried enabling both the ss_redir and ss_server section?  From the screenshot you gave, you need to click button Enable or 启用

MisterLin1995, 2018-02-27:
I tried to start ss-redir in command line and 
the shadowsocks-libev-3.1.3 will encounter： error loading shared library libmbedcrypto.so.0.
Finally I solved this problem by uninstalling
 all other packages
 and only using shadowsocks-libev-ss-redir-3.1.3-2.

MisterLin1995, 2018-02-27:
@yousong And the button in screenshot means it is already enabled，pressing it only cause the pid to get killed.

yousong, 2018-02-27:
Guess I will need remote access to see the exact cause like previously done with another issue (https://github.com/openwrt/packages/issues/4614#issuecomment-317261405).  Can you contact me by email and setup the session ready?

MisterLin1995, 2018-02-27:
I cannot access my router now since I'm not going back in 4 months. You can email me so that I will be able to send you logs or other materials to locate the key of the problems.

MisterLin1995, 2018-02-27:
I guess the key problem lies in my router firmware because it's a Snapshot version or there's something wrong with  shadowsocks-libev-3.1.3  since shadowsocks-libev-ss-redir-3.1.3-2 worked.

yousong, 2018-02-27:
@MisterLin1995 , Please post the content of `/etc/config/shadowsocks-libev` with server hostname and password stripped off.  Please also share where the firmware was got and how you customised it as the libmbedcrypt.so error should not happen with vanilla OpenWrt build.

MisterLin1995, 2018-02-27:
I have no access to that router now since I didn't open the remote access.
I bought the router on taobao.com and the firmware is built in so I have no source firmware. 
I only uninstalled all pre-installed packages installed by the seller. And then I followed the instruction of https://cokebar.info/archives/664 . 
The arch of the router cpu should be  mips 24Kc if I remember it correctly.
And the router model is wndr4300 V2, not wndr4500 V3  as shown in the scrennshot.
I can provide the purchase link if you email me so I can reply that to you privately.

yousong, 2018-02-27:
Well, now I am in a "works for me" situation.  Please keep in mind that current packaging of shadowsocks-libev in this repo is a complete rewrite.  The link you posted above is very likely obsolete in some aspects.  You can find my email address at [1] when you are ready

[1] https://github.com/openwrt/packages/blob/master/net/shadowsocks-libev/Makefile#L23

MisterLin1995, 2018-02-27:
Got it,thank you very much!

closed 2018-02-27 by MisterLin1995