openwrt/packages #5652: cloud flare ddns v4 subdomain support

gxgani, 2018-02-21:
ddns-scripts_cloudflare.com-v4

# Issue template (remove lines from top till here)

Maintainer: @chris5560 
Environment: WRT3200ACM

Description:

```
Would it be possible to implement subdomain support for cloudflare dns? I am unable to figure out the api call.
```

gxgani, 2018-02-21:
found it via #5097.

closed 2018-02-21 by gxgani

disrupted, 2019-02-08:
did you get it to work for multiple subdomains? for me currently it only updates the A record for mydomain.tld, not subdomain1.mydomain.tld, subdomain2.mydomain.tld and so on

gxgani, 2019-02-09:
> 
> 
> did you get it to work for multiple subdomains? for me currently it only updates the A record for mydomain.tld, not subdomain1.mydomain.tld, subdomain2.mydomain.tld and so on

Yea, make sure you have ddns-scripts and ddns-scripts-cloudflarev4 is installed. This is how my config looks

[image]

disrupted, 2019-02-10:
Mine looks very similar to yours but only works for the main domain for now. what did you enter for `subdomain` if you want it to update for all of your subdomains? the wildcard `*`?
also can I ask which version of OpenWRT are you running? just noticed your LuCI looks so different to mine.