openwrt/packages #5665: Transmission can't connect to SSL trackers

SqTER-Pila, 2018-02-22:
After one of the latest source updates there is a problem with Transmission. Unable to connect to SSL trackers.
Checked transmission-daemon-mbedtls-2.93-4 and transmission-daemon-openssl-2.93-4
System: OpenWrt SNAPSHOT r6073-49d3c5f
Device: NETGEAR WNDR 3800 router
Kernel: 4.9.77 # 0 Mon Feb 12

dibdot, 2018-02-26:
@neheb

neheb, 2018-02-26:
I turned on verification for TLS certificates. Probably a self signed certificate is the issue. Do you have an example?

SqTER-Pila, 2018-02-26:
Certificate issuer  Adguard Personal CA
The exported certificate is attached
The problem concerns a few pages with a similar certificate

[cert.zip](https://github.com/openwrt/packages/files/1760165/cert.zip)

neheb, 2018-02-26:
I meant an example tracker but that works. I'll look at it later today.

Are you using mbedtls or OpenSSL?

SqTER-PL, 2018-02-26:
Trackers are private. I can not share the address
I am currently using Transmission 2.92 mbedtls on lede-17.01 but the OpenSSL also works well
The problem is on the openwrt trunk

neheb, 2018-02-26:
The easiest way to fix this is to change this line to 0 in /etc/init.d/transmission
```procd_set_param env TR_CURL_SSL_VERIFY=1```

The certificate failed verification on Fedora. It's possible that ca-bundle needs to be updated. I will look into this.

SqTER-PL, 2018-02-27:
I check it out tonight and let you know.

SqTER-PL, 2018-02-27:
Your fix does not work, but if I use this file ( https://github.com/openwrt/packages/blob/lede-17.01/net/transmission/files/transmission.init ) it works fine.

neheb, 2018-02-27:
The only other fix I can imagine is to delete both env lines completely.

SqTER-PL, 2018-02-27:
I deleted two env lines and the problem was solved.
Everything is working properly.

neheb, 2018-03-01:
I googled this Adguard thing and found that it's software for HTTPS filtering...

Sorry, but I'm not reverting this change.

SqTER-PL, 2018-03-01:
I use AdGuard so I could send a bad certificate
Please check these pages (https://polishsource.cz , https://btgigs.info). Encrypted connections to these addresses do not work.
In my opinion, the certificates are fine

neheb, 2018-03-01:
Ok so some other issue. Will investigate.

I personally don't have any torrents that support HTTPS trackers so I haven't been able to test. Not sure if there are public ones that do.

SqTER-PL, 2018-03-01:
You can create an account on Polishsource.

neheb, 2018-03-02:
I just tested this. TLS verification is totally broken from the looks of it. I changed the port test from HTTP to HTTPS and it stopped working. Reverting until this is fixed.

SqTER-PL, 2018-03-02:
It's great that you found the cause of the problem. Let me know how everything will work well. I'll check it.

neheb, 2018-03-02:
New version has been merged. Please close if no more issues.

SqTER-PL, 2018-03-02:
Solved
Thanks :-)

closed 2018-03-02 by SqTER-PL