openwrt/packages #5692: perl: Broken build with PIE enabled

neheb, 2018-03-01:
Maintainer: @pprindeville 

It's failing here: ```make[5]: Leaving directory '/home/mangix/devstuff/openwrt/build_dir/target-mips_74kc_musl/perl/perl-5.26.1/cpan/DB_File'
make[5]: Entering directory '/home/mangix/devstuff/openwrt/build_dir/target-mips_74kc_musl/perl/perl-5.26.1/cpan/DB_File'
/home/mangix/devstuff/openwrt/build_dir/target-mips_74kc_musl/perl/perl-5.26.1/cpan/DB_File/../../../../../../staging_dir/hostpkg/usr/bin/perl "-I../../lib" -MExtUtils::Command::MM -e 'cp_nonempty' -- DB_File.bs ../../lib/auto/DB_File/DB_File.bs 644
rm -f ../../lib/auto/DB_File/DB_File.so
LD_RUN_PATH="/home/mangix/devstuff/openwrt/staging_dir/target-mips_74kc_musl/usr/lib" mips-openwrt-linux-musl-ld  -shared -L/home/mangix/devstuff/openwrt/staging_dir/target-mips_74kc_musl/usr/lib -L/home/mangix/devstuff/openwrt/staging_dir/target-mips_74kc_musl/lib -L/home/mangix/devstuff/openwrt/staging_dir/toolchain-mips_74kc_gcc-5.5.0_musl/usr/lib -L/home/mangix/devstuff/openwrt/staging_dir/toolchain-mips_74kc_gcc-5.5.0_musl/lib -specs=/home/mangix/devstuff/openwrt/include/hardened-ld-pie.specs -znow -zrelro version.o DB_File.o  -o ../../lib/auto/DB_File/DB_File.so  \
   -ldb   \
  
mips-openwrt-linux-musl-ld: unrecognized option '-specs=/home/mangix/devstuff/openwrt/include/hardened-ld-pie.specs'
mips-openwrt-linux-musl-ld: use the --help option for usage information
make[5]: *** [Makefile:501: ../../lib/auto/DB_File/DB_File.so] Error 1
make[5]: Leaving directory '/home/mangix/devstuff/openwrt/build_dir/target-mips_74kc_musl/perl/perl-5.26.1/cpan/D```

For some reason earlier in the build, it builds stuff using cc instead off gcc. I don't think the host compiler should be being used. Example: ```make[5]: Entering directory '/home/mangix/devstuff/openwrt/build_dir/hostpkg/perl/perl-5.26.1/ext/re'
rm -f re_exec.c
cp ../../regexec.c re_exec.c
rm -f invlist_inline.h
cp ../../invlist_inline.h invlist_inline.h
cc -c   -fwrapv -fno-strict-aliasing -pipe -fstack-protector-strong -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -D_FORTIFY_SOURCE=2 -Wall -Werror=declaration-after-statement -Wextra -Wc++-compat -Wwrite-strings -O2   -DVERSION=\"0.34\" -DXS_VERSION=\"0.34\"  "-I../.."  -DPERL_EXT_RE_BUILD -DPERL_EXT_RE_DEBUG -DPERL_EXT re_exec.c
re_exec.c: In function 'S_regmatch':
re_exec.c:6742:2: warning: this statement may fall through [-Wimplicit-fallthrough=]
  {
  ^
re_exec.c:7130:2: note: here
  case EVAL_postponed_AB: /* cleanup after a successful (??{A})B */
  ^~~~
re_exec.c:7398:2: warning: this statement may fall through [-Wimplicit-fallthrough=]
  {
  ^
re_exec.c:7429:2: note: here
  case CURLYX_end: /* just finished matching all of A*B */
  ^~~~
re_exec.c:8299:9: warning: this statement may fall through [-Wimplicit-fallthrough=]
         {
         ^
re_exec.c:8344:2: note: here
  case IFMATCH_A_fail: /* body of (?...A) failed */
  ^~~~
rm -f re_comp.c
cp ../../regcomp.c re_comp.c
rm -f dquote.c
cp ../../dquote.c dquote.c
cc -c   -fwrapv -fno-strict-aliasing -pipe -fstack-protector-strong -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -D_FORTIFY_SOURCE=2 -Wall -Werror=declaration-after-statement -Wextra -Wc++-compat -Wwrite-strings -O2   -DVERSION=\"0.34\" -DXS_VERSION=\"0.34\"  "-I../.."  -DPERL_EXT_RE_BUILD -DPERL_EXT_RE_DEBUG -DPERL_EXT re_comp.c
"../../miniperl" "-I../../lib" "../../lib/ExtUtils/xsubpp" -noprototypes -typemap '/home/mangix/devstuff/openwrt/build_dir/hostpkg/perl/perl-5.26.1/ext/re/../../lib/ExtUtils/typemap'  re.xs > re.xsc
mv re.xsc re.c
cc -c   -fwrapv -fno-strict-aliasing -pipe -fstack-protector-strong -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -D_FORTIFY_SOURCE=2 -Wall -Werror=declaration-after-statement -Wextra -Wc++-compat -Wwrite-strings -O2   -DVERSION=\"0.34\" -DXS_VERSION=\"0.34\"  "-I../.."  -DPERL_EXT_RE_BUILD -DPERL_EXT_RE_DEBUG -DPERL_EXT re.c
rm -f "../../lib/auto/re/re.a"
/usr/bin/ar cr "../../lib/auto/re/re.a" re_exec.o re_comp.o re.o
: "../../lib/auto/re/re.a"
chmod 755 ../../lib/auto/re/re.a
make[5]: Leaving directory '/home/mangix/devstuff/openwrt/build_dir/hostpkg/perl/perl-5.26.1/ext/re'
```

Masanetz, 2018-03-30:
Worked around this problem by adding
PKG_ASLR_PIE:=0
to the perl Makefile...

pprindeville, 2018-03-30:
That should only be used for compilers that don't understand ASLR hardening, not for all architectures.

Also, ASLR wouldn't change which compiler was being used.

@neheb There are 2 binaries built, including a host binary called `miniperl` which is then used to boot-strap the build process, since Perl is used to build Perl...  `miniperl` should be built with the host compiler.

I can't tell what's going on from the information you've provided above.

I'd need the relevant parts of your `.config` at a minimum (stripped of anything extraneous) to reproduce this.

ZeroChaos-, 2018-04-23:
builds fine like this:

# CONFIG_PKG_ASLR_PIE is not set
CONFIG_GCC_DEFAULT_PIE=y
# CONFIG_BUSYBOX_DEFAULT_PIE is not set
CONFIG_BUSYBOX_CONFIG_PIE=y

pprindeville, 2018-04-23:
Why are busybox changes necessary to get Perl to build properly?

neheb, 2018-04-23:
No idea. Now that I think about it, CONFIG_PKG_ASLR_PIE is enabled for me but I can't reproduce. CONFIG_BUSYBOX_DEFAULT_PIE is not set.

ZeroChaos-, 2018-04-23:
I suspected that the busybox change was tied to the other one, I didn't toggle it manually, I only manually toggled CONFIG_PKG_ASLR_PIE with "Global build settings -> User space ASLR PIE compilation"

ZeroChaos-, 2018-04-23:
make sure you "make dirclean" when changing a setting like this to ensure the toolchain is built properly with the desired setting.

pzskc383, 2018-06-19:
Happens to me too on openwrt-18.06  brannch.
```
CONFIG_PKG_ASLR_PIE=y
CONFIG_GCC_DEFAULT_PIE=y
CONFIG_BUSYBOX_DEFAULT_PIE=y
CONFIG_BUSYBOX_CONFIG_PIE=y
```

cshoredaniel, 2018-07-30:
@pprindeville PIE may not be the culprit.  Perl is failing to build with the warning implicit-fall-through on Fedora 28.  This may be related to *host* version of GCC (see core for CFLAGS that have been used to deal with 'overly aggressive' warnings from GCC 8).

ZeroChaos-, 2018-07-30:
I have had this issue with host on gcc 6.x and gcc 7.x

neheb, 2018-07-30:
Last I remember this issue it was on Ubuntu 14.04 WSL.

pprindeville, 2018-07-30:
Most of my builds are on Ubuntu 14.04 LTS (_trusty tahr_) but give me a few minutes to build on F28.

pprindeville, 2018-07-30:
I'm getting read to update to 5.28.0 which resolves the `Archive::Tar` CVE.

Has anyone else tried building that?

cshoredaniel, 2018-07-31:
I'm planning on trying now that you point it out, probably tomorrow.

cshoredaniel, 2018-08-02:
And now a different build is working ... I hate non-deterministic failures....

pprindeville, 2018-08-02:
@cshoredaniel Sorry, not following.  Did you rebase and pick up my fixes?

cshoredaniel, 2018-08-02:
Hmmmm.....actually this was a master build and the other was 18.06, that is probably what's different.   18.06 FTBFS on Fedora 28, but master is okay....I'll crawl back under my rock now...

pprindeville, 2018-08-03:
@cshoredaniel Just updated 18.06.

pprindeville, 2018-08-07:
Closing for now

closed 2018-08-07 by pprindeville