openwrt/packages #5693: i2pd: create_directory: Permission denied

dartraiden, 2018-03-01:
> daemon.err i2pd[8266]:   what():  boost::filesystem::create_directory: Permission denied: "/var/lib/i2pd/destinations"
> daemon.err i2pd[8302]:   what():  boost::filesystem::create_directory: Permission denied: "/var/lib/i2pd/tags"
> 

Seems like /etc/init.d/i2pd should contain something like
`chown i2pd:i2pd /var/lib/i2pd`
otherwise /var/lib/i2pd belongs to _root_ and i2pd cannot create necessary directories

dartraiden, 2018-03-01:
Also problem with certificates:

[image]

but i2pd with _DATADIR=/etc/i2pd_ search them into /etc/i2pd/.i2pd

dibdot, 2018-03-01:
@yangfl

dartraiden, 2018-03-03:
So we have 2 ways:

1) keep --service in i2pd.init, give i2pd write permissions to /var/lib/i2pd and make ln -s /usr/share/i2pd/certificates /var/lib/i2pd/certificates

pros: working directory is /var/lib/i2pd = no flash memry wearing
cons: not all routers have enough memory

2) remove --service from init, and somehow teach i2pd use /etc/i2pd instead of /etc/i2pd/.i2pd/

yangfl, 2018-03-03:
We should at least keep peer profiles persistent, shouldn't we?

LLE8, 2018-03-03:
No. Files are permanently overwritten and they take large spase. Flash memory will die. RAM only.

dartraiden, 2018-03-03:
On my router (256 MB RAM) works perfectly:

```
#!/bin/sh /etc/rc.common
# Copyright (C) 2017 OpenWrt.org

USE_PROCD=1

START=90
STOP=10

PROG=/usr/sbin/i2pd
USER="i2pd"
GROUP="i2pd"
PIDFILE=/var/run/i2pd.pid
DATADIR=/var/lib/i2pd
CERTDIR=$DATADIR/certificates
TUNCONF=$DATADIR/tunnels.conf

start_service() {
	if [ ! -d $CERTDIR ]; then
		ln -s /usr/share/i2pd/certificates $CERTDIR
	fi
	if [ ! -d $TUNCONF ]; then
		ln -s /etc/i2pd/tunnels.conf $TUNCONF
	fi

	## We need permissions
	chown -R $USER:$GROUP $DATADIR
	touch $PIDFILE
	chown $USER:adm $PIDFILE

	procd_open_instance
	procd_set_param command $PROG --service --conf=/etc/i2pd/i2pd.conf
	## Don't know about i2pd user's HOME
	procd_set_param env HOME=$DATADIR
	procd_set_param limits nofile=4096
	procd_set_param stdout 1
	procd_set_param stderr 1
	procd_set_param user $USER
	procd_set_param pidfile $PIDFILE
	procd_close_instance
}
```

yangfl, 2018-03-03:
Maybe link to /etc/i2pd/addressbook ?
It's not possible to me to use http reseeding, I have to manually import addressbook.

dartraiden, 2018-03-03:
you mean, store addressbok into /etc/i2pd?

yangfl, 2018-03-03:
Yes, and link corresponding files from /var/lib to /etc. Others LGFM and if you can't wait you can create a pr for it.

dartraiden, 2018-03-03:
Done.

Makefile:
```
#
# Copyright (C) 2015, 2016 gxcreator
# Copyright (C) 2017 OpenWrt.org
#
# This is free software, licensed under the GNU General Public License v2.
# See /LICENSE for more information.
#

include $(TOPDIR)/rules.mk

PKG_NAME:=i2pd
PKG_VERSION:=2.18.0
PKG_RELEASE:=1
PKG_BUILD_PARALLEL:=1

PKG_SOURCE_PROTO:=git
PKG_SOURCE:=$(PKG_NAME)-$(PKG_VERSION).tar.gz
PKG_MIRROR_HASH:=1a04826b54e649ebb341d3feacdd05c0415335679d1aa787a74ea5995d419378
PKG_SOURCE_SUBDIR:=$(PKG_NAME)-$(PKG_VERSION)
PKG_SOURCE_URL:=https://github.com/PurpleI2P/i2pd.git
PKG_SOURCE_VERSION:=$(PKG_VERSION)
PKG_LICENSE:=BSD-3-clause

include $(INCLUDE_DIR)/package.mk

define Package/i2pd
	SECTION:=net
	CATEGORY:=Network
	DEPENDS:=+libopenssl +boost +boost-system +boost-filesystem \
	         +boost-program_options +boost-date_time +libatomic +zlib
	TITLE:=full-featured C++ implementation of I2P client
	URL:=https://github.com/PurpleI2P/i2pd
	USERID:=i2pd:i2pd
	MAINTAINER:=David Yang <mmyangfl@gmail.com>
endef

define Package/i2pd/description
	I2P (Invisible Internet Protocol) is a universal anonymous network layer. All
	communications over I2P are anonymous and end-to-end encrypted, participants
	don't reveal their real IP addresses.
endef

define Package/i2pd/conffiles
	/etc/i2pd/i2pd.conf
	/etc/i2pd/subscriptions.txt
	/etc/i2pd/tunnels.conf
endef

TARGET_LDFLAGS+=-latomic
MAKE_FLAGS+=USE_AESNI=no USE_AVX=no

define Package/i2pd/install
	$(INSTALL_DIR) $(1)/usr/sbin
	$(INSTALL_BIN) $(PKG_BUILD_DIR)/i2pd $(1)/usr/sbin
	$(INSTALL_DIR) $(1)/usr/share/i2pd
	$(CP) $(PKG_BUILD_DIR)/contrib/certificates  $(1)/usr/share/i2pd
	$(INSTALL_DIR) $(1)/etc/i2pd
	$(INSTALL_BIN) $(PKG_BUILD_DIR)/contrib/i2pd.conf $(1)/etc/i2pd
	$(SED) ' \
		s/127.0.0.1/192.168.1.1/g; \
		s/datadir = \/var\/lib/datadir = \/etc/ \
	' $(1)/etc/i2pd/i2pd.conf
	$(INSTALL_BIN) $(PKG_BUILD_DIR)/contrib/subscriptions.txt $(1)/etc/i2pd
	$(INSTALL_BIN) $(PKG_BUILD_DIR)/contrib/tunnels.conf $(1)/etc/i2pd
	$(INSTALL_DIR) $(1)/etc/init.d
	$(INSTALL_BIN) ./files/i2pd.init $(1)/etc/init.d/i2pd
endef

$(eval $(call BuildPackage,i2pd))
```

i2pd.init:
```
#!/bin/sh /etc/rc.common
# Copyright (C) 2017 OpenWrt.org

USE_PROCD=1

START=90
STOP=10

PROG=/usr/sbin/i2pd
USER="i2pd"
GROUP="i2pd"
PIDFILE=/var/run/i2pd.pid
DATADIR=/var/lib/i2pd

start_service() {
	## RAM
	if [ ! -d $DATADIR ]; then
		mkdir -p $DATADIR
		ln -s /usr/share/i2pd/certificates  $DATADIR/certificates
		ln -s /etc/i2pd/tunnels.conf $DATADIR/tunnels.conf
		# for peoples who not possible to use http reseeding
		ln -s /etc/i2pd/addressbook $DATADIR/addressbook
	fi
	
	## We need permissions
	chown $USER:$GROUP $DATADIR
	touch $PIDFILE
	chown $USER:adm $PIDFILE

	procd_open_instance
	procd_set_param command $PROG --service --conf=/etc/i2pd/i2pd.conf --pidfile $PIDFILE
	## Don't know about i2pd user's HOME
	procd_set_param env HOME=$DATADIR
	procd_set_param limits nofile=4096
	procd_set_param stdout 1
	procd_set_param stderr 1
	procd_set_param user $USER
	procd_set_param pidfile $PIDFILE
	procd_close_instance
}
```

commit 87a7ec72 (2020-02-07): i2pd: Move DATADIR to /var/lib/i2pd, fix #5693

Original author: dartraiden <wowemuh@gmail.com>