openwrt/packages #5720: mpd-full or mpd-mini doesn't build (armeb_xscale_musl)

trurle, 2018-03-07:
Maintainer: @thess 
Environment: Intel IXP4xx-generic, armeb_xscale_musl, OpenWRT master (b562c0c91b0cfb7a02a7ca69063bebda6225a672 at the time of writing)
Host: Debian 8.7 (Linux debian 3.16.0-4-amd64 #1 SMP Debian 3.16.39-1 (2016-12-30) x86_64 GNU/Linux)

```
CFLAGS="-Os -pipe -mcpu=xscale -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -iremap/home/muxx/devel/openwrt/build_dir/target-armeb_xscale_musl/mpd-mini/mpd-0.20.9:mpd-0.20.9 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/muxx/devel/openwrt/staging_dir/target-armeb_xscale_musl/usr/lib/libiconv-stub/include -I/home/muxx/devel/openwrt/staging_dir/target-armeb_xscale_musl/usr/lib/libintl-stub/include  -I/home/muxx/devel/openwrt/staging_dir/target-armeb_xscale_musl/usr/include -I/home/muxx/devel/openwrt/staging_dir/target-armeb_xscale_musl/include -I/home/muxx/devel/openwrt/staging_dir/toolchain-armeb_xscale_gcc-5.5.0_musl/usr/include -I/home/muxx/devel/openwrt/staging_dir/toolchain-armeb_xscale_gcc-5.5.0_musl/include/fortify -I/home/muxx/devel/openwrt/staging_dir/toolchain-armeb_xscale_gcc-5.5.0_musl/include -I/home/muxx/devel/openwrt/staging_dir/target-armeb_xscale_musl/usr/lib/libiconv-stub/include -I/home/muxx/devel/openwrt/staging_dir/target-armeb_xscale_musl/usr/lib/libintl-stub/include " CXXFLAGS="-Os -pipe -mcpu=xscale -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -iremap/home/muxx/devel/openwrt/build_dir/target-armeb_xscale_musl/mpd-mini/mpd-0.20.9:mpd-0.20.9 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/muxx/devel/openwrt/staging_dir/target-armeb_xscale_musl/usr/lib/libiconv-stub/include -I/home/muxx/devel/openwrt/staging_dir/target-armeb_xscale_musl/usr/lib/libintl-stub/include  -std=gnu++14 -I/home/muxx/devel/openwrt/staging_dir/target-armeb_xscale_musl/usr/include -I/home/muxx/devel/openwrt/staging_dir/target-armeb_xscale_musl/include -I/home/muxx/devel/openwrt/staging_dir/toolchain-armeb_xscale_gcc-5.5.0_musl/usr/include -I/home/muxx/devel/openwrt/staging_dir/toolchain-armeb_xscale_gcc-5.5.0_musl/include/fortify -I/home/muxx/devel/openwrt/staging_dir/toolchain-armeb_xscale_gcc-5.5.0_musl/include -I/home/muxx/devel/openwrt/staging_dir/target-armeb_xscale_musl/usr/lib/libiconv-stub/include -I/home/muxx/devel/openwrt/staging_dir/target-armeb_xscale_musl/usr/lib/libintl-stub/include " LDFLAGS="-L/home/muxx/devel/openwrt/staging_dir/target-armeb_xscale_musl/usr/lib -L/home/muxx/devel/openwrt/staging_dir/target-armeb_xscale_musl/lib -L/home/muxx/devel/openwrt/staging_dir/toolchain-armeb_xscale_gcc-5.5.0_musl/usr/lib -L/home/muxx/devel/openwrt/staging_dir/toolchain-armeb_xscale_gcc-5.5.0_musl/lib -znow -zrelro -L/home/muxx/devel/openwrt/staging_dir/target-armeb_xscale_musl/usr/lib/libiconv-stub/lib -Wl,-rpath-link=/home/muxx/devel/openwrt/staging_dir/target-armeb_xscale_musl/usr/lib/libiconv-stub/lib -L/home/muxx/devel/openwrt/staging_dir/target-armeb_xscale_musl/usr/lib/libintl-stub/lib -Wl,-rpath-link=/home/muxx/devel/openwrt/staging_dir/target-armeb_xscale_musl/usr/lib/libintl-stub/lib -Wl,--whole-archive -liconv -Wl,--no-whole-archive" make  -C /home/muxx/devel/openwrt/build_dir/target-armeb_xscale_musl/mpd-mini/mpd-0.20.9/. AR="armeb-openwrt-linux-muslgnueabi-gcc-ar" AS="armeb-openwrt-linux-muslgnueabi-gcc -c -Os -pipe -mcpu=xscale -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -iremap/home/muxx/devel/openwrt/build_dir/target-armeb_xscale_musl/mpd-mini/mpd-0.20.9:mpd-0.20.9 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/muxx/devel/openwrt/staging_dir/target-armeb_xscale_musl/usr/lib/libiconv-stub/include -I/home/muxx/devel/openwrt/staging_dir/target-armeb_xscale_musl/usr/lib/libintl-stub/include" LD=armeb-openwrt-linux-muslgnueabi-ld NM="armeb-openwrt-linux-muslgnueabi-gcc-nm" CC="armeb-openwrt-linux-muslgnueabi-gcc" GCC="armeb-openwrt-linux-muslgnueabi-gcc" CXX="armeb-openwrt-linux-muslgnueabi-g++" RANLIB="armeb-openwrt-linux-muslgnueabi-gcc-ranlib" STRIP=armeb-openwrt-linux-muslgnueabi-strip OBJCOPY=armeb-openwrt-linux-muslgnueabi-objcopy OBJDUMP=armeb-openwrt-linux-muslgnueabi-objdump SIZE=armeb-openwrt-linux-muslgnueabi-size CROSS="armeb-openwrt-linux-muslgnueabi-" ARCH="armeb" ;
make[4]: Entering directory '/home/muxx/devel/openwrt/build_dir/target-armeb_xscale_musl/mpd-mini/mpd-0.20.9'
make  all-am
make[5]: Entering directory '/home/muxx/devel/openwrt/build_dir/target-armeb_xscale_musl/mpd-mini/mpd-0.20.9'
armeb-openwrt-linux-muslgnueabi-g++ -DHAVE_CONFIG_H -I.  -DNDEBUG -I./src  -DSYSTEM_CONFIG_FILE_LOCATION='"/etc/mpd.conf"' -I/home/muxx/devel/openwrt/staging_dir/target-armeb_xscale_musl/usr/include    -I/home/muxx/devel/openwrt/staging_dir/target-armeb_xscale_musl/usr/include -I/home/muxx/devel/openwrt/staging_dir/target-armeb_xscale_musl/include -I/home/muxx/devel/openwrt/staging_dir/toolchain-armeb_xscale_gcc-5.5.0_musl/usr/include -I/home/muxx/devel/openwrt/staging_dir/toolchain-armeb_xscale_gcc-5.5.0_musl/include/fortify -I/home/muxx/devel/openwrt/staging_dir/toolchain-armeb_xscale_gcc-5.5.0_musl/include -I/home/muxx/devel/openwrt/staging_dir/target-armeb_xscale_musl/usr/lib/libiconv-stub/include -I/home/muxx/devel/openwrt/staging_dir/target-armeb_xscale_musl/usr/lib/libintl-stub/include  -D_GNU_SOURCE -pthread -Os -pipe -mcpu=xscale -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -iremap/home/muxx/devel/openwrt/build_dir/target-armeb_xscale_musl/mpd-mini/mpd-0.20.9:mpd-0.20.9 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/muxx/devel/openwrt/staging_dir/target-armeb_xscale_musl/usr/lib/libiconv-stub/include -I/home/muxx/devel/openwrt/staging_dir/target-armeb_xscale_musl/usr/lib/libintl-stub/include  -std=gnu++14 -fvisibility=hidden -fno-threadsafe-statics -fmerge-all-constants -ffast-math -ftree-vectorize -ffunction-sections -fdata-sections -Wall -Wextra -Wmissing-declarations -Wshadow -Wpointer-arith -Wcast-qual -Wwrite-strings -Wsign-compare -MT src/output/libmpd_a-Internal.o -MD -MP -MF src/output/.deps/libmpd_a-Internal.Tpo -c -o src/output/libmpd_a-Internal.o `test -f 'src/output/Internal.cxx' || echo './'`src/output/Internal.cxx
In file included from src/output/Internal.cxx:21:0:
src/output/Internal.hxx:266:7: error: 'exception_ptr' in namespace 'std' does not name a type
  std::exception_ptr last_error;
       ^
src/output/Internal.hxx:319:13: error: 'exception_ptr' in namespace 'std' does not name a type
  const std::exception_ptr &GetLastError() const {
             ^
Makefile:9673: recipe for target 'src/output/libmpd_a-Internal.o' failed
make[5]: *** [src/output/libmpd_a-Internal.o] Error 1
make[5]: Leaving directory '/home/muxx/devel/openwrt/build_dir/target-armeb_xscale_musl/mpd-mini/mpd-0.20.9'
Makefile:4681: recipe for target 'all' failed
make[4]: *** [all] Error 2
make[4]: Leaving directory '/home/muxx/devel/openwrt/build_dir/target-armeb_xscale_musl/mpd-mini/mpd-0.20.9'
Makefile:212: recipe for target '/home/muxx/devel/openwrt/build_dir/target-armeb_xscale_musl/mpd-mini/mpd-0.20.9/.built' failed
make[3]: *** [/home/muxx/devel/openwrt/build_dir/target-armeb_xscale_musl/mpd-mini/mpd-0.20.9/.built] Error 2
make[3]: Leaving directory '/home/muxx/devel/openwrt/feeds/packages/sound/mpd'
package/Makefile:106: recipe for target 'package/feeds/packages/mpd/compile' failed
make[2]: *** [package/feeds/packages/mpd/compile] Error 2
make[2]: Leaving directory '/home/muxx/devel/openwrt'
package/Makefile:102: recipe for target '/home/muxx/devel/openwrt/staging_dir/target-armeb_xscale_musl/stamp/.package_compile' failed
make[1]: *** [/home/muxx/devel/openwrt/staging_dir/target-armeb_xscale_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/muxx/devel/openwrt'
```
[dotconfig.txt](https://github.com/openwrt/packages/files/1787234/dotconfig.txt)

thess, 2018-03-07:
I have been painfully aware of this problem since last Dec. as you can see here:
https://github.com/lede-project/source/pull/1572#issuecomment-350919129

I was hoping we would have moved onto GCC 7 by now - maybe soon. If you really need to build MPD for an ARM5/6, I think my patch is around somewhere. It makes a serious mod to how C++ exceptions are handled by the entire system. MPD did build and function with this patch but I would not recommend using the patched GCC for any other platform builds.

trurle, 2018-03-12:
it builds and functions ok with:
CONFIG_GCC_VERSION_7=y
CONFIG_GCC_VERSION="7.3.0"

thess, 2018-03-13:
Good to know it works - I knew it built OK but haven't had the time to actually test it on an IXP4xx system yet.

thess, 2018-03-13:
GCC 7 and BINUTILS 2.29.1 are now default for most targets. Closing this.

closed 2018-03-13 by thess