openwrt/packages #5721: tmux: doesn't show up on the console

alive4ever, 2018-03-07:
Maintainer: @diizzyy
Environment: x86_64, qemu, snapshot r6395-a70db57

Description:
When launching tmux directly on the OpenWrt console or serial, tmux doesn't show up.
Input works, but there is no tmux panel shown at all.

[image]

mstorchak, 2018-03-07:
Maybe console doesn't provide all capabilities tmux requires.
How should the fix look like?

alive4ever, 2018-03-19:
The fix may involves fixing [procd](https://git.openwrt.org/project/procd.git) console initialization to add capabilities tmux needs.

alive4ever, 2018-03-26:
comparing output of stty on the openwrt console to actual linux console, the openwrt console output lacks the following capabilities:
```
-cmspar
-flusho
-extproc
```

alive4ever, 2018-03-26:
procd seems to lack console initialization to set terminal configuration.
busybox init does have [console_init](https://git.busybox.net/busybox/tree/init/init.c#n296).

neheb, 2020-01-12:
I assume this is still an issue.

mstorchak, 2020-01-12:
Is it a procd issue then?