openwrt/packages #5728: <node>: Illegal instruction with raspberry pi zero w

godunkchen, 2018-03-08:
Hi

  Got "Illegal instruction" while executing node.
  From gdb trace, it shows fault at instruction

=> 0x36e24dd8:  movw    r7, #53100      ; 0xcf6c

```
root@OpenWrt:~/sandbox# cat test.js
console.log("hello world!");
root@OpenWrt:~/sandbox# node test.js
Illegal instruction
```

Environment: (put here arch, model, OpenWrt version)
root@OpenWrt:~/sandbox# uname -a
Linux OpenWrt 4.9.85 #0 Tue Mar 6 07:52:13 2018 armv6l GNU/Linux
root@OpenWrt:~/sandbox# cat /proc/cpuinfo
processor       : 0
model name      : ARMv6-compatible processor rev 7 (v6l)
BogoMIPS        : 697.95
Features        : half thumb fastmult vfp edsp java tls
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xb76
CPU revision    : 7

Hardware        : BCM2835
Revision        : 9000c1
Serial          : 00000000ae06c113

Description:

root@OpenWrt:~/sandbox# gdb node test.js
GNU gdb (GDB) 8.0.1
Copyright (C) 2017 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "arm-openwrt-linux".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from node...(no debugging symbols found)...done.
"/root/sandbox/test.js" is not a core dump: File format not recognized
(gdb) run
Starting program: /usr/bin/node
warning: Unable to find dynamic linker breakpoint function.
GDB will be unable to debug shared library initializers
and track explicitly loaded dynamic code.
[New LWP 4693]
[New LWP 4694]
[New LWP 4695]
[New LWP 4696]
[New LWP 4697]

Thread 1 "node" received signal SIGILL, Illegal instruction.
0x36e24dd8 in ?? ()
(gdb) bt
#0  0x36e24dd8 in ?? ()
#1  0x36e200e8 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) disassemble $pc,$pc+32
Dump of assembler code from 0x36e24dd8 to 0x36e24df8:
=> 0x36e24dd8:  movw    r7, #53100      ; 0xcf6c
   0x36e24ddc:  movt    r7, #155        ; 0x9b
   0x36e24de0:  ldr     r7, [r7]
   0x36e24de4:  movw    r10, #49188     ; 0xc024
   0x36e24de8:  movt    r10, #155       ; 0x9b
   0x36e24dec:  push    {r1}            ; (str r1, [sp, #-4]!)
   0x36e24df0:  push    {r2}            ; (str r2, [sp, #-4]!)
   0x36e24df4:  ldr     r2, [r10, #1416]        ; 0x588
End of assembler dump.

```
Formating code blocks by wrapping them with pairs of ```
```

nxhack, 2018-03-09:
Probably...
https://chromium.googlesource.com/v8/v8/+/886b259e229572257e6271c3f6d1532c510e15ef%5E%21/

Update for "chromium-v8" is done in node v7.x upper.

nxhack, 2018-03-09:
Tested with Raspbian Stretch and node.js official node-v6.11.2-linux-armv6l, it worked normally. There may be problems from OpenWrt something.

nxhack, 2018-03-20:
Raspbian:
```
raspberrypi:~ $ echo | gcc -dM -E - | grep ARM_ARCH
#define __ARM_ARCH_ISA_ARM 1
#define __ARM_ARCH_6__ 1
#define __ARM_ARCH_ISA_THUMB 1
#define __ARM_ARCH 6
```
OpenWrt:
```
ubuntu:~ $ $ echo | ./arm-openwrt-linux-muslgnueabi-gcc -dM -E - | grep ARM_ARCH
#define __ARM_ARCH_ISA_ARM 1
#define __ARM_ARCH_5T__ 1
#define __ARM_ARCH_ISA_THUMB 1
#define __ARM_ARCH 5
```

nxhack, 2018-03-23:
Probably we can fix this.
https://github.com/openwrt/packages/pull/5800

nxhack, 2018-07-10:
@godunkchen 
The package has been updated.
Is the problem solved?

godunkchen, 2018-07-13:
@nxhack 
Sorry for late.
I've just verified it from upgrading node to version v8.11.3-1, it works fine now.

Great thanks for your contribution.

> root@OpenWrt:~# opkg info node
> Package: node
> Version: v8.11.3-1
> Depends: libc, libstdcpp, libopenssl, zlib
> Status: install user installed
> Section: lang
> Architecture: arm_arm1176jzf-s_vfp
> Size: 4462403
> Filename: node_v8.11.3-1_arm_arm1176jzf-s_vfp.ipk
> Description: Node.js® is a JavaScript runtime built on Chrome's V8 JavaScript engine. Node.js uses
>  an event-driven, non-blocking I/O model that makes it lightweight and efficient. Node.js'
>  package ecosystem, npm, is the largest ecosystem of open source libraries in the world.
> Installed-Time: 1531453633
> 

```
root@OpenWrt:~# cat test.js
console.log("hello world!");
```

> root@OpenWrt:~# node test.js
> hello world!

closed 2018-07-13 by godunkchen